corna/me_cleaner #71: Asus Z10PE-D8 WS (motherboard); no FTPR modules except BUP?

sakaki-, 2017-10-03:
Hi,

Thanks for making this tool available!

I have run `me_cleaner` successfully now on three PCs now, two laptops and a server (reports [here](https://github.com/corna/me_cleaner/issues/3#issuecomment-333393692), [here](https://github.com/corna/me_cleaner/issues/3#issuecomment-333503394) and [here](https://github.com/corna/me_cleaner/issues/3#issuecomment-333811917)).

On the two laptop systems, `me_cleaner` produced a long list of FTPR modules that were removed (kernel, policy etc.). However, on the server, an ASUS Z10PE-D8 WS motherboard, only BUP was detected (and not removed). Output of `me_cleaner` run copied again below:

```console
root@pi64 ~ # python ../me_cleaner/me_cleaner.py -S original6.rom -O modified6.rom
Full image detected
The ME/TXE region goes from 0x1000 to 0x800000
Found FPT header at 0x1010
Found 6 partition(s)
Found FTPR header: FTPR partition spans from 0x22000 to 0x42000
ME/TXE firmware version 3.1.3.30
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x4b)...
Reading FTPR modules list...
 BUP              (uncomp., 0x02235c - 0x03935c): NOT removed, essential
The ME minimum size should be 249856 bytes (0x3d000 bytes)
The ME region can be reduced up to:
 00001000:0003dfff me
Setting the AltMeDisable bit in PCHSTRP10 to disable Intel ME...
Checking the FTPR RSA signature... VALID
Done! Good luck!

root@pi64 ~ # cmp -l original6.rom modified6.rom | wc -l
1048321
```

Is that what would be expected given this is a server rather than a consumer machine BIOS? At any event the ME _appears_ to have been disabled (see [report](https://github.com/corna/me_cleaner/issues/3#issuecomment-333811917)), and `me_cleaner` has written about 1MiB of 0xFF bytes into the ROM image.

corna, 2017-10-03:
Hi,

you're trying to disable a SPS firmware, which is [officially not supported](https://github.com/corna/me_cleaner/wiki/me_cleaner-status) (yet). Seems to work well in your case though.

Thanks for providing me the link and commands needed to get the rom file (instead of sending me the extracted file, which is not allowed).

From the log I see that the FTPR partition contains only a partition (not removed), however there were 6 partitions, 5 of which have been removed. The ME region is huge (8 MB) however the SPS firmware occupies only the first 1.5 MB, which is not much compared to the other Intel firmware, so I suppose this one is a minimal firmware and it seems resonable that the FTPR partition contains only the BUP module. @platomav is probably more informed than me.

Also consider that _me_cleaner_ removed most of the code, from ~1.5 MB to ~90 kB (but it didn't show it, as it wasn't in the FTPR partition).

Can you try again without the `-s` flag and report the results? No need to tell you that this might cause a brick.

platomav, 2017-10-03:
SPS is built from two modules: Recovery (FTPR) and Operational (OPR). Everything else is generated automatically by Intel's Flash Image Tool. For redundancy, SPS can have Backup regions which are usually copies of the original (MFS & MFSB, OPR1 & OPR2). Little is known about SPS and even less people have systems to test but the general idea is that OPR loads first, if it fails it goes to backup and if that fails, it loads FTPR. OPR can be configured to work either as a simple platform configurator (Silicon Enabling mode) or with Node Manager application (Node Manager mode).

For SPS 3 only, the $FPT has a [2-byte checksum](https://github.com/platomav/MEAnalyzer/blob/86dded490dd8882a7c6dd5205ff778a3c2412b4f/MEA.py#L3360) field at the end of its entries which me_cleaner does not adjust currently.

[image]

Also, the total Engine region size does not matter because, by default, FIT appends all remaining SPI image padding to the Engine region.

So, other than the SPS3-specific checksum, I don't see anything wrong with me_cleaner's actions.

sakaki-, 2017-10-03:
@corna:
> Can you try again without the -s flag and report the results? No need to tell you that this might cause a brick.

Happy to do so. This is a production machine though, so it'll probably be the weekend before I can try it.

sakaki-, 2017-10-09:
OK just retried without `-S` flag. Very similar results. Survives 30mins, some BIOS state lost (drops back to legacy boot initially, EFI boot list has been forgotten when EFI boot re-enabled), but that happens with the -S image too (I have migrated to the more modern 3407 BIOS).

Here is the `me_cleaner` output _with_ `-S` (on the 3407 BIOS rom):

```
Full image detected
The ME/TXE region goes from 0x1000 to 0x800000
Found FPT header at 0x1010
Found 6 partition(s)
Found FTPR header: FTPR partition spans from 0x22000 to 0x42000
ME/TXE firmware version 3.1.3.38
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x4b)...
Reading FTPR modules list...
 BUP              (uncomp., 0x02235c - 0x03935c): NOT removed, essential
The ME minimum size should be 249856 bytes (0x3d000 bytes)
The ME region can be reduced up to:
 00001000:0003dfff me
Setting the AltMeDisable bit in PCHSTRP10 to disable Intel ME...
Checking the FTPR RSA signature... VALID
Done! Good luck!
```

and _without_ `-S`:

```
Full image detected
The ME/TXE region goes from 0x1000 to 0x800000
Found FPT header at 0x1010
Found 6 partition(s)
Found FTPR header: FTPR partition spans from 0x22000 to 0x42000
ME/TXE firmware version 3.1.3.38
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x4b)...
Reading FTPR modules list...
 BUP              (uncomp., 0x02235c - 0x03935c): NOT removed, essential
The ME minimum size should be 249856 bytes (0x3d000 bytes)
The ME region can be reduced up to:
 00001000:0003dfff me
Checking the FTPR RSA signature... VALID
Done! Good luck!
```

Here is the `intelmetool` output _with_ `-S`:

```
MEI not hidden on PCI, checking if visible
MEI found: [8086:8d3a] C610/X99 series chipset MEI Controller #1

ME Status   : 0xf0382
ME Status 2 : 0x9040d008

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : YES
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Recovery
ME: Current Operation State : Bring up
ME: Current Operation Mode  : M0 without UMA but with error
ME: Error Code              : No Error
ME: Progress Phase          : Uncategorized Failure
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : Unknown 0x40

ME: Extend Register not valid

ME: timeout waiting for data: expected 8, available 6
ME: GET FW VERSION message failed
ME: timeout waiting for data: expected 5, available 0
ME: GET FWCAPS message failed
```

and _without_ `-S`:

```
MEI was hidden on PCI, now unlocked
MEI found: [8086:8d3a] C610/X99 series chipset MEI Controller #1

ME Status   : 0xf0382
ME Status 2 : 0x90401008

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : YES
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Recovery
ME: Current Operation State : Bring up
ME: Current Operation Mode  : M0 without UMA but with error
ME: Error Code              : No Error
ME: Progress Phase          : Uncategorized Failure
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : Unknown 0x40

ME: Extend Register not valid

ME: timeout waiting for data: expected 8, available 6
ME: GET FW VERSION message failed
ME: timeout waiting for data: expected 5, available 0
ME: GET FWCAPS message failed
Re-hiding MEI device...done
```

Diff between `-S` and no `-S` outputs of `intelmetool`:

```
1c1
< MEI not hidden on PCI, checking if visible
---
> MEI was hidden on PCI, now unlocked
5c5
< ME Status 2 : 0x9040d008
---
> ME Status 2 : 0x90401008
26a27
> Re-hiding MEI device...done
```

Does that mean the AltMeDisable bit is being ignored on SPS firmware?

hth, sakaki

corna, 2017-10-13:
Thanks for the tests.

Yes, it seems that AltMeDisable is ignored on SPS (but the modules removal works on your platform at least).

closed 2017-10-13 by corna