corna/me_cleaner #72: Create v1.1 release?

sakaki-, 2017-10-07:
Hi,
I'd like to add an ebuild for `me_cleaner` to my [64-bit Gentoo RPi3 overlay](https://github.com/sakaki-/rpi3-overlay) - would it be possible to create a v1.1 release of `me_cleaner` please? I can setup the ebuild to clone a particular commit, but it is better to use GitHub's release tarballs where possible (clients don't require `git` installed that way). Thanks!

evadim, 2017-10-07:
To be honest, to add your overlay user need to instal git anyway - it hosted at github. Your sentence only correct if ebuild added to main tree.

sakaki-, 2017-10-07:
Well true, although I am hoping to upstream some of the ebuilds shortly, hence the request.
Also release-based ebuilds are easier to revbump.
But in any event, I've added a commit-based ebuild for `me_cleaner` ([here](https://github.com/sakaki-/rpi3-overlay/blob/master/sys-apps/me_cleaner/me_cleaner-1.0.20171007.ebuild)) pro tem.

corna, 2017-10-08:
Hi,

I'm going to release the 2.0 version soon, just after
 * reviewing and merging #70
 * adding more comments to the code (based on #57)
 * testing extensively the code

sakaki-, 2017-10-08:
Great, I'll create a release-based ebuild for this when you push it.

Also, I have just posted a draft issue of my [EFI Install Guide](https://wiki.gentoo.org/wiki/Sakaki%27s_EFI_Install_Guide) addendum, covering use of your `me_cleaner` tool, [**here**](https://wiki.gentoo.org/index.php?title=Sakaki%27s_EFI_Install_Guide/Disabling_the_Intel_Management_Engine).

I based it loosely on the flow in your [External flashing](https://github.com/corna/me_cleaner/wiki/External-flashing) wiki page. If you get a chance to have a quick look over it to see if there are any major errors, prior to me linking it into the guide, it'd be much appreciated. Thanks!

corna, 2017-10-10:
Great guide! I've added it to the [How to apply me_cleaner](https://github.com/corna/me_cleaner/wiki/How-to-apply-me_cleaner) page.

sakaki-, 2017-10-11:
Thanks ^-^
Just noticed it got picked up on [ycombinator](https://news.ycombinator.com/item?id=15444607), quite a long thread already, so hopefully will encourage a few more people to try `me_cleaner`!

corna, 2017-10-13:
A friend of mine sent it to me, only after reading it I realized you were the author. :wink:

I hope so!

corna, 2018-01-02:
Sorry for the **huuuuuge** delay, [here](https://github.com/corna/me_cleaner/releases/tag/v1.1) it is.

closed 2018-01-02 by corna

sakaki-, 2018-01-05:
Thanks, I've just pushed a 1.1 ebuild for this.

sakaki-, 2018-01-25:
FYI, I've just pushed a v1.2.0 of the `gentoo-on-rpi3-64bit` bootable image to GitHub ([here](https://github.com/sakaki-/gentoo-on-rpi3-64bit)). This now comes bundled with v1.1 of `me_cleaner`, plus `flashrom`, `ifdtool` and `wiringpi` (for the `gpio` command) pre-installed. The SPI interface is also enabled by default. So users wanting to use their RPi3 as an external flashing tool just need to burn the image, boot (the image starts up automatically to an Xfce4 destkop), connect the right clip and go. No Gentoo knowledge or software installation necessary ^-^