corna/me_cleaner #74: Does updating BIOS overwrites me_cleaner's work?

kjrew, 2017-10-17:
If I update BIOS does it overwrite what me_cleaner have done?
Or does it update just the UEFI's region and doesn't touch ME region?

corna, 2017-10-17:
Likely, many BIOS updates also contains an Intel ME image, both to update or restore a corrupted ME firmware.

kjrew, 2017-10-17:
Any way to check if it contains it before updating?

skochinsky, 2017-10-19:
You can try searching for the `$FPT` signature in the BIOS update, though if you don't find one it does *not* mean it hasn't one (e.g. it could be in a compressed form).

Yes, Lenovo updates usually don't touch the ME (but I guess exceptions are always possible).

kjrew, 2017-10-19:
ok, thanks!

closed 2017-10-19 by kjrew

corna, 2017-10-19:
UEFI updates are usually compressed, so a string search won't probably give you any result. Luckily Lenovo usually provides good changelogs with its BIOS updates, so you can easily see if they included a ME update.