corna/me_cleaner #75: ASUS Z170-P (CPU i7-6700). Is switched really -off?

arsonoff, 2017-10-17:
motherboard ASUS Z170-P (CPU i7-6700)
before BIOS change, I have:
```
# lspci -v
...
00:16.0 Communication controller: Intel Corporation Sunrise Point-H CSME HECI #1 (rev 31)
    Subsystem: ASUSTeK Computer Inc. Device 8694
    Flags: bus master, fast devsel, latency 0, IRQ 16
    Memory at f722d000 (64-bit, non-prefetchable) [size=4K]
    Capabilities: [50] Power Management version 3
    Capabilities: [8c] MSI: Enable- Count=1/1 Maskable- 64bit+
    Kernel modules: mei_me
...
# intelmetool -s
MEI not hidden on PCI, checking if visible
MEI found: [8086:a13a] Sunrise Point-H CSME HECI #1

ME Status   : 0x90000245
ME Status 2 : 0x66000506

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : YES
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Normal
ME: Current Operation State : M0 with UMA
ME: Current Operation Mode  : Normal
ME: Error Code              : No Error
ME: Progress Phase          : Host Communication
ME: Power Management Event  : Pseudo-global reset
ME: Progress Phase State    : Host communication established

ME: Extend Register not valid

ME: failed to become ready
ME: failed to become ready
ME: GET FW VERSION message failed
ME: failed to become ready
ME: failed to become ready
ME: GET FWCAPS message failed
```
When using the utility without additional options or with an option `-S` I receive **not operable** BIOS.
Output of command `./me_cleaner.py -S -O z170-p.noME-S.bin z170-p.bin`:
```
Full image detected
The ME/TXE region goes from 0x1000 to 0x280000
Found FPT header at 0x1010
Found 11 partition(s)
Found FTPR header: FTPR partition spans from 0x2000 to 0xa9000
Found FTPR manifest at 0x2478
ME/TXE firmware version 11.0.0.1168
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x06)...
Reading FTPR modules list...
 FTPR.man     (uncompressed, 0x002478 - 0x00307c): NOT removed, partition manif.
 rbe.met      (uncompressed, 0x00307c - 0x003112): NOT removed, module metadata
 kernel.met   (uncompressed, 0x003112 - 0x0031a0): NOT removed, module metadata
 syslib.met   (uncompressed, 0x0031a0 - 0x003204): NOT removed, module metadata
 bup.met      (uncompressed, 0x003204 - 0x00368a): NOT removed, module metadata
 pm.met       (uncompressed, 0x00368a - 0x003738): NOT removed, module metadata
 syncman.met  (uncompressed, 0x003738 - 0x0037ce): NOT removed, module metadata
 vfs.met      (uncompressed, 0x0037ce - 0x004122): NOT removed, module metadata
 evtdisp.met  (uncompressed, 0x004122 - 0x0042b0): NOT removed, module metadata
 loadmgr.met  (uncompressed, 0x0042b0 - 0x0043d8): NOT removed, module metadata
 busdrv.met   (uncompressed, 0x0043d8 - 0x004788): NOT removed, module metadata
 gpio.met     (uncompressed, 0x004788 - 0x004894): NOT removed, module metadata
 prtc.met     (uncompressed, 0x004894 - 0x004a44): NOT removed, module metadata
 policy.met   (uncompressed, 0x004a44 - 0x004c0c): NOT removed, module metadata
 crypto.met   (uncompressed, 0x004c0c - 0x004d96): NOT removed, module metadata
 heci.met     (uncompressed, 0x004d96 - 0x004f32): NOT removed, module metadata
 storage.met  (uncompressed, 0x004f32 - 0x00520a): NOT removed, module metadata
 pmdrv.met    (uncompressed, 0x00520a - 0x00532e): NOT removed, module metadata
 maestro.met  (uncompressed, 0x00532e - 0x005412): NOT removed, module metadata
 fpf.met      (uncompressed, 0x005412 - 0x005506): NOT removed, module metadata
 hci.met      (uncompressed, 0x005506 - 0x0056b0): NOT removed, module metadata
 fwupdate.met (uncompressed, 0x0056b0 - 0x0057b8): NOT removed, module metadata
 ptt.met      (uncompressed, 0x0057b8 - 0x0058aa): NOT removed, module metadata
 touch_fw.met (uncompressed, 0x0058aa - 0x0059c0): NOT removed, module metadata
 rbe          (Huffman     , 0x0059c0 - 0x008ac0): NOT removed, essential
 kernel       (Huffman     , 0x008ac0 - 0x017a80): NOT removed, essential
 syslib       (Huffman     , 0x017a80 - 0x02a680): NOT removed, essential
 bup          (Huffman     , 0x02a680 - 0x050c00): NOT removed, essential
 pm           (LZMA/uncomp., 0x050c00 - 0x053580): removed
 syncman      (LZMA/uncomp., 0x053580 - 0x053ac0): removed
 vfs          (LZMA/uncomp., 0x053ac0 - 0x05b700): removed
 evtdisp      (LZMA/uncomp., 0x05b700 - 0x05d180): removed
 loadmgr      (LZMA/uncomp., 0x05d180 - 0x05ff00): removed
 busdrv       (LZMA/uncomp., 0x05ff00 - 0x062bc0): removed
 gpio         (LZMA/uncomp., 0x062bc0 - 0x064040): removed
 prtc         (LZMA/uncomp., 0x064040 - 0x064dc0): removed
 policy       (LZMA/uncomp., 0x064dc0 - 0x06b200): removed
 crypto       (LZMA/uncomp., 0x06b200 - 0x07ac40): removed
 heci         (LZMA/uncomp., 0x07ac40 - 0x07ecc0): removed
 storage      (LZMA/uncomp., 0x07ecc0 - 0x083340): removed
 pmdrv        (LZMA/uncomp., 0x083340 - 0x084b00): removed
 maestro      (LZMA/uncomp., 0x084b00 - 0x087740): removed
 fpf          (LZMA/uncomp., 0x087740 - 0x0890c0): removed
 hci          (LZMA/uncomp., 0x0890c0 - 0x089940): removed
 fwupdate     (LZMA/uncomp., 0x089940 - 0x08e380): removed
 ptt          (LZMA/uncomp., 0x08e380 - 0x0a2180): removed
 touch_fw     (LZMA/uncomp., 0x0a2180 - 0x0a9000): removed
The ME minimum size should be 344064 bytes (0x54000 bytes)
The ME region can be reduced up to:
 00001000:00054fff me
Setting the HAP bit in PCHSTRP0 to disable Intel ME...
Checking the FTPR RSA signature... VALID
Done! Good luck!
```
When using the utility with an option `-s` I receive **operable** BIOS.
Output of command `./me_cleaner.py -s -O z170-p.noME-s.bin z170-p.bin`:
```
Full image detected
The ME/TXE region goes from 0x1000 to 0x280000
Found FPT header at 0x1010
Found 11 partition(s)
Found FTPR header: FTPR partition spans from 0x2000 to 0xa9000
Found FTPR manifest at 0x2478
ME/TXE firmware version 11.0.0.1168
Setting the HAP bit in PCHSTRP0 to disable Intel ME...
Checking the FTPR RSA signature... VALID
Done! Good luck!
```
at the same time the `00:16.0 Communication controller:...` device vanishes (is absent in output of command `lspci`), output of command `intelmetool-s`:
```
MEI not hidden on PCI, checking if visible
MEI device not found
```
And should be? How to check that `Intel ME` is switched really -off?

mparnelldmp, 2017-10-23:
If it has dual bios chips you may have to flash both...and with some bioses you need to set the reflash me boolean to true, using AMIBCP if it's an AMI bios.

If you set that boolean, then you need to flash twice to actually trigger the ME flash.

corna, 2017-10-23:
`MEI device not found` is a good indication that ME is switched off. Unfortunately Sunrise Point is not supported yet by _intelmetool_, so you can't check properly if it is really disabled. There is an Intel tool to check the status of Intel ME but, unfortunately, it is proprietary and not redistributable.

It is quite uncommon however that it works with `-s` but not with `-S`.

arsonoff, 2017-10-24:
>There is an Intel tool to check the status of Intel ME 
```
Intel(R) MEInfo Version: 11.0.26.3000
Copyright(C) 2005 - 2017, Intel Corporation. All rights reserved.
    
    Table Type   0 ( 0x 00 ) found, size of  24 (0x 18 ) bytes
...
    Table Type 219 ( 0x DB ) found, size of  81 (0x 51 ) bytes
FW Status Register1: 0x80022004
FW Status Register2: 0x364D0006
FW Status Register3: 0x00000000
FW Status Register4: 0x00086000
FW Status Register5: 0x00000000
FW Status Register6: 0x40000000

  CurrentState:                         Disabled
  ManufacturingMode:                    Disabled
  FlashPartition:                       Valid
  OperationalState:                     Transitioning
  InitComplete:                         Initializing
  BUPLoadState:                         Success
  ErrorCode:                            Disabled
  ModeOfOperation:                      Alt Disable Mode
  SPI Flash Log:                        Not Present
  Phase:                                BringUp
  ICC:                                  Valid OEM data, ICC programmed
  ME File System Corrupted:             No
  PhaseStatus:                          UNKNOWN
  
  FPF and ME Config Status:             Match
  
Error 86: Communication error between application and Intel(R) ME module (FWU client)
  
Error 81: Internal error (Could not determine FW features information)
```

closed 2017-10-24 by arsonoff