corna/me_cleaner #76: Results with late 2011 MacBook Pro 13 inch

matt123b, 2017-10-21:
I wanted to report success with this laptop. It has a Core i5 2435M. I had to use a Raspberry Pi to dump the ROM and then I ran ME Cleaner on the copied firmware and then flashed it back. After nearly 10 minutes (thought I bricked it, lol) of power cycling it finally came to life. The screen was still connected and the screen lit up but was still showing black. Then it beeped a few times and the fan sped up but then I saw an Apple logo indicating that the Apple EFI was working. I then plugged in a hard disk with OS X Sierra on it and it booted up just fine. Then I live booted a Ubuntu flash drive and the Intel ME no longer shows up in the PCI devices. I would test it on more Macs but my newer ones are for work and I don't want to risk them. But for anyone else that's more brave than I am, this does appear to work on Apple MacBooks. 

I have also tried this on an extra ThinkPad T400 (Core2 P8400) with the stock Lenovo BIOS and it works great, though it's kind of pointless since Libreboot exists. I will look around the house for more Intel Core i machines that aren't mission critical and I'll test this on those too if I can.

matt123b, 2017-10-21:
This is the exact laptop, by the way. Just in case anyone wanted to know. 

https://support.apple.com/kb/sp645?locale=en_US
https://everymac.com/systems/apple/macbook_pro/specs/macbook-pro-core-i5-2.4-13-late-2011-unibody-thunderbolt-specs.html

corna, 2017-10-22:
Great news!!

What do you mean with "After nearly 10 minutes of power cycling it finally came to life."? Did you power cycle it once and waited for 10 minutes, or did you continuously power cycling it?

matt123b, 2017-10-22:
I would plug the battery and power adapter in and turn it on and the fan would spin at about 1/4 speed and the little light near the front of the machine's palm rest would glow but there was no activity otherwise. So I would wait a few seconds and then unplug and start over. After about 10 minutes of that it finally worked. I knew not to give up because I experienced something similar when installing Coreboot and Libreboot on some of my other laptops.

corna, 2017-10-22:
Wonderful!

Can you write a small guide about it, something like [this one](https://github.com/corna/me_cleaner/wiki/me_cleaner-on-a-MSI-H110M-ECO)?

corna, 2017-10-22:
Which arguments did you use with me_cleaner?

youssef-lr, 2017-10-23:
It should work on latest macbook pros too?

corna, 2017-10-23:
I think (and hope) so, someone should test it.

likeaross, 2017-10-30:
I think the safest way to test and modify MacBook firmware would be to use something like the [Matt Card](http://www.cmizapper.com/products/mattcard.html)

matt123b, 2017-10-30:
"Can you write a small guide about it, something like this one?"

I don't think that it's really necessary since the guide in the wiki is what I used to do this. I followed all of the instructions and just Googled to find out how to do anything I didn't know how to do, such as find the chip model and stuff. I used this entry in the wiki here:

https://github.com/corna/me_cleaner/wiki/External-flashing

I skipped over the "Neutralize and shrink Intel ME (useful only for coreboot)" section obviously because I'm running the Apple EFI. I also didn't bother with the intelmetool. I first tried booting OS X off the main hard disk with worked fine, and then I hooked up another hard disk with regular old Ubuntu LTS and pressed Option until I got the EFI bootable disks menu and selected it. Then I ran lspci and the ME was gone.

matt123b, 2017-10-30:
"I think the safest way to test and modify MacBook firmware would be to use something like the Matt Card"

Huh, I never thought about doing something like that. That's pretty smart and convenient.

chaosbunker, 2017-10-30:
Can anyone explain how exactly a Matt Card would help disable the Intel ME? Doesn't it just bypass the EFI Lock? And what is running on the matt card? and why is there a slot for it in the first place?

matt123b, 2017-10-30:
"Can anyone explain how exactly a Matt Card would help disable the Intel ME?"

The Apple EFI and the Intel ME firmware are stored on the same chip. So if the Matt Card is already programmed then it would contain a full known good image dumped from the ROM on another Mac of the same model. You could also program it yourself in the same way if it comes empty. 

"Doesn't it just bypass the EFI Lock?"

It could do that, yes. But the ME firmware is what initializes the hardware so if you're only booting from the Matt Card ROM you would need ME firmware present which is bundled in the normal dumps anyways. 

"And what is running on the matt card?"

Everything that would be on the soldered in ROM chip, so the Apple EFI and ME firmware basically. 

"and why is there a slot for it in the first place?"

Probably for manufacturing, testing, flashing the original internal ROM? Maybe it's left over from R&D and it wouldn't be useful or profitable to modify the board designs? Who knows.

nightyasen, 2017-11-13:
Is the Matt Card suitable for permanent use?
I remember a problem with the old xbox mod chips eventually burning out 1.6 xboxs.

matt123b, 2017-11-15:
I doubt it. I have no idea how the boot process of an Xbox works though. If the Matt Card thing does wear out, you'll probably have a new machine by then. Otherwise it's just 60 euro for a new one, which isn't nothing but it's not that horribly expensive, especially when dealing with Macs which are some of the most overpriced machines on the face of the Earth.

corna, 2017-11-15:
@nightyasen from what I see the Matt card is just a flash chip with a proprietary connector, perfectly suitable for permanent use.

HazCod, 2017-11-22:
Is the Macbook relevant, as the Apple EFI does not use Intel ME at all?

platomav, 2017-11-22:
All Intel-based systems have the Engine co-processor and thus CS(ME)/CS(TXE)/CS(SPS) firmware. That obviously includes Apple.

johnkeates, 2017-11-26:
The diagnostic port on the Mac motherboards are for manufacturing and debugging. The pins include SPI and LPC, but it has power and SPI flash select override. This means that you can plug in a compatible SPI flash chip and take the SPI select line on the connector and instruct the SMC and PCH to use the external SPI instead of the on-board one. 

While this has been used to bypass iCloud firmware locks, it basically is no different than swapping the on-board SPI flash or it's contents.

The mainboards also have Mini XDP ports for the PCH and CPU, which are standard Intel processor and PCH debugging systems. They are not fitted with a connector, but are otherwise intact and usable.

The motherboard schematics on the LPC+SPI Connector (this is actually what it's called on the A1278 schematics PDF) has the line named SPIROM_USE_MLB, which is connected to GPIO57 on the cougar point PCH and to #HOLD on the SST25VF064C 64Mbit SPI flash. R2193 pulls it to PP3V3_SUS using a 100K resistor by default making sure the on-board SPI flash is not on-hold. If you then pull the SPIROM_USE_MLB pin to GND on the LPC+SPI Connector, it puts the on-board SPI flash in HOLD mode ignoring all SPI cycles.

The ME firmware in Apple EFI is often rather old (9.x on the 2015MBP I have here atm) and other than basic drivers for power management it's not really used by Apple's OS.

corna, 2017-11-26:
@johnkeates thanks for the detailed explanation.

johnkeates, 2017-11-26:
Oh, and also: this best works when using a modified dump of the existing SPI contents since it does contain stuff like the MAC address, serial number, board serial (yes, different numbers apparently) etc. as well as some security settings and iCloud membership status. If you keep it blank, the macOS will probably freak out, and if you put in random stuff, things like iCloud, iMessage etc. will not work. I'm not sure if the SMC firmware is completely in there as well, but that's a rather important embedded controller in Mac computers. If it doesn't work properly, the machine won't start, or will work slow or really buggy.

So far, the ME_Cleaner used on downloadable Apple EFI firmware updates does work fine, so I suspect that as long as you use it on the firmware you already have in the on-board SPI it will be fine. Something else to keep in mind is the fact that Apple supplies it's firmware updates using the on-board AppStore/softwareupdate mechanism in macOS. While you can cancel them, I don't think there is an alternative way to install firmware updates. The firmware packages come with a macOS-only installer that dumps some stuff on the EFI partition and then the EFI bootmanager will start it. I suppose if you get rEFInd on there and an EFI shell, you could update it manually, which eliminates the need for SPI flashing.

matt123b, 2017-11-30:
@HazCod Intel recommends that manufacturers include the ME firmware. As far as Apple and other companies have been told, the ME firmware itself is not optional, and that only some modules and configurations are optional. Even if Apple were to strip the firmware, the ME could have reliability issues or have trouble updating to patch vulnerabilities. So Apple has left the ME firmware alone. And since the ME firmware is stored on the BIOS ROM, it needs read/write access to that ROM to update. Because of that, your BIOS (Apple EFI in this case) should be considered compromised due to a lack of isolation. And there's also the problem with the ME interfacing with all of the other devices, including your hard disks. It's truly a security disaster, and Apple's EFI doesn't even have to touch it. 

@johnkeates I like the idea of using an alternative boot manager to open a shell to try to install updates. I just did an OS update on the MacBook and it worked fine, though I didn't record the previous EFI version, so I have no clue whether it updated. I'll be sure to write down this one and compare it when I update next.