corna/me_cleaner #77: OEM Firmware only available in exe; how to use me_cleaner with this?

z0m8i3, 2017-10-26:
From the [wiki](https://github.com/corna/me_cleaner/wiki/Internal-flashing-with-OEM-firmware) I got as far as:

> Now check if the firmware has the correct structure:
> $ ifdtool -d original_firmware.bin

I am running a Lenovo Ideapad and the firmware is only available in .exe format on Lenovo's site; how do I obtain a .bin?

I couldn't find any steps on the web regarding how to clean the exe and/or generate a .bin if one isn't provided by the manufacturer.

Anyone else encounter this or have any options how to get around it?

Thanks to all who've participated in developing and testing me_cleaner!

corna, 2017-10-26:
You can try with binwalk, but I'm afraid you won't get anything interesting. Unfortunately, if the vendor doesn't provide a "plain" firmware image, the only alternative is to use an [external programmer](https://github.com/corna/me_cleaner/wiki/External-flashing). Moreover I think the Lenovo flashing tool does signature checking on their firmwares, so you would have needed an external programmer anyways to flash back the modified ME image.

skochinsky, 2017-10-26:
innoup/innoextract may work on Lenovo installers, but AFAIK they usually only include the BIOS region updates, so you will probably need an external programmer to access the ME region anyway.

z0m8i3, 2017-10-27:
Unfortunate news!
External flashing is beyond my current level of expertise (even with the how-to), I will have to get my hands on some test devices to get more comfortable with the process before I proceed with my main PC.

Thanks for the replies.

closed 2017-10-27 by z0m8i3