corna/me_cleaner #78: Info about used NSA ports?

JurekRaben, 2017-10-26:
In a world with a working jurisdiction, you could sue NSA to death, because they tampering with your personal security. And sue Intel to remove it from all computers immediately and also pay lot of compensation. But sadly, such system does not exist. Even worse, NSA was widely hacked months ago ("They even don't know what was stolen"). 

So we now have to assume that not only NSA has access to all of our computers thru this Intel ME network stack, but also FSB, Mafia and other criminals. Even more sad that people and politicians don't care, because they cannot imagine or are simply dumb. Literally they can hack everything from us, and we won't know. They even can destroy our lives and we wouldn't know the reason for that.

That's why a deep analysis of that network stack would be interesting. What ports are used? How can we block it using a open source firmware router? Is there any information about that? Thanks.

If there were criminals around your neighbourhood having the key for your home, you would secure your home by changing the lock, or add another lock.

matt123b, 2017-10-30:
The problem with "deep state" government agencies like the NSA/CIA is that they operate outside of the law. Even if you could get your day in court, these claims of their involvement in it is purely conjecture. If anything it's Intel that's screwing us over. But they have a monopoly on x86 hardware (and they'll sue over compatibility layers like when they threatened MS not to make ARM laptops with certain features in the x86 compatibility layer). At least their management engine has some documentation and we know what it is since the remote access "features" are advertised to enterprise customers, whereas the AMD PSP is an undocumented blackbox. The only reason why I even buy Intel workstation hardware is because I'm more comfortable with the evil I know than the one I don't. That and I'm too poor for a Talos II.

mastad0n, 2017-12-28:
@JurekRaben @matt123b I think its port 16992 and 16993, but I'm not nearly as advanced in any of this as you guys are. I found it here:

https://software.intel.com/sites/manageability/AMT_Implementation_and_Reference_Guide/default.htm?turl=WordDocuments%2Faccessingintelamtviathewebuiinterface.htm