corna/me_cleaner #79: Experiance with HP Elitebook 8470p; Will not flash, OEM tool method

Null-Senpai, 2017-10-29:
Admittedly I don't fully understand a lot of this, but here is how far I have gotten on a HP Elitebook 8470p.

**Info**
`BIOS Date: 07/13/2017`
`System BIOS Version: 68ICF Ver. F.68`

Intel ME version should be 8.1.71.3608, as that is the newest available and I updated it not too long ago due to CVE-2017-5689, although I did not find a way to check the installed version.

**Process**
Although `ifdtool` did not really recognize the image, I was fairly certain that the .bin file provided by HP is not some weird special format, so I tried `me_cleaner -c original.bin` and it mentioned that it was indeed a ME image, as well as that the "FTPR RSA signature is valid". I assume I won't be able to flash things without a valid signature.

After producing the image (no special parameters, just the standard `me_cleader.py -O mod.bin original.bin`)  and rebooting, trying to flash it using the UEFI file results in an error:
`Error 8771: Invalid File`.

I assumed maybe the signature is no longer valid, but running `me_cleaner -c mod.bin` still showed the signature to be valid.

**Possible Alternative**
Since I am not sure what the issue is with trying to flash the modded ME.bin, I was considering using the NSA-way and just having the HAP thing enabled, however this requires flashing the full bios not just the ME part I believe. Unfortunately I have not found a way to flash the bios without using Microsoft Windows, so I gave up here.

mparnelldmp, 2017-10-30:
Most modern HP machines have a form of self protection where an OEM backup bios gets flashed if a modified one is flashed due to the RSA signature involved. Did the stock bios you modified include a .sig file?

If so, you'll need to use AMIBCP to toggle the ME reflash bit to on, and then flash the bios twice via SPI externally, unless you can get AFUDOS or AFULINUX to flash it properly.

Afterward HP's recovery may still forcibly reflash their own bios, I haven't yet researched disabling the recovery/RSA key check. I just know that in my situation that even when this happened, that because I had previously flashed a clean ME and the version numbers matched, the default bios didn't update the ME and so it remained disabled.

Null-Senpai, 2017-10-30:
For my Laptop at least, HP provides the two separately, the BIOS itself (which I did not investigate much) and the Intel ME update.
I only modified the latter (me_cleaner calls it a "ME/TXE image") and tried to flash it with the included UEFI program. It did not have any .sig files.
I might have a look at AMIBCP, however I don't think the issue was the device restoring the ME, rather [I was unable to flash it at all (using the OEM tool)](https://user-images.githubusercontent.com/13616397/32182247-1c316c52-bd8e-11e7-99d9-cb91d9444a5c.jpg). I tried flashing the original ME, and even though it's the same version, it did proceed to update so I don't think it's refusing because it's the same version. 

I will also have a look at the other tools, but I fear my only option may be external flashing, something I am not that comfortable doing since it'd be the first time and I would have no idea what I am doing.

mparnelldmp, 2017-10-30:
Yes, for my laptop it is the same, and the process I describe will apply there.

Yes, the OEM tool is probably not going to work. AFU (dos/win/linux) will be more likely to flash. There's also an EFI version too. External flashing in my opinion is in many cases easier than doing a dance with OEM tools, unless you have to take apart a laptop whose chip isn't just right there when you take off the bottom panel.

Null-Senpai, 2017-10-30:
I wouldn't mind the idea of flashing it externally, especially since it seems the ~~best~~ only way to do a proper backup and not brick my device, but it's obviously not as easy as following instructions for software. I don't even know what the chip looks like physically 😄 
Next weekend I will have a look at AFU and UEFITool, since HP's own end-user software is not gonna get me far it seems.

If everything fails I might just buy [one of those SPI flashers](https://www.ebay.com/itm/CH341A-Series-SPI-Flasher-USB-Programmer-24-25-EEPROM-BIOS-Writer-USB-to-TTL/381206013212)  and see how far I get. Thankfully the device is pretty easy to service so I doubt I will have any issues getting to the chip once I find out where it is.

matt123b, 2017-10-30:
I would always recommend using an external flashing device because it won't have any issues reading or writing the ROM like internal flashing sometimes does. I know that taking apart your computer isn't really quick or convenient but you'll be able to easily restore everything if you mess it up the first time and it should be a one time process for most machines. Seriously, even if internal flashing is available, just do it externally. It's more straightforward in many ways and it'll save you a lot of headaches.

Null-Senpai, 2017-10-30:
It's not really the taking apart, but more the 'Will I have to randomly unsolder chips from my motherboard' part. 
That said, I will order the SPI flasher since it's cheap and I am a bit more afraid now, since it seems that pretty much anywhere I look the general consensus is 'don't do it using software'.
So just to clarify, flashing it externally, I can theoretically flash anything I want onto the device, and there will be nothing preventing me from actually flashing it?

mparnelldmp, 2017-10-30:
If your chip is removable it's easy, if soldered on, it's still mostly easy if you get a proper pomona or knockoff soic clip.

The only dance there is figuring out if you need to bring up a couple pins or not, or whether your machine should be plugged in or not...depends on the machine, there, I think. It's even different between thinkpads on that one.

matt123b, 2017-10-31:
"So just to clarify, flashing it externally, I can theoretically flash anything I want onto the device, and there will be nothing preventing me from actually flashing it?"

Pretty much, yeah. But there's no guarantee that it'll boot up and work normally after and you might have to restore to the original, unmodified image that you dumped before flashing the stripped version. 

"It's not really the taking apart, but more the 'Will I have to randomly unsolder chips from my motherboard' part."

What, what? There shouldn't be any soldering involved if you're the one flashing. I mean you could solder on a larger chip if you wanted more space for Coreboot and big payloads or something that's pre-flashed but you normally just hook up a clip that will allow you to read/write the ROM chip and it's really easy to do using a Raspberry Pi running Debian where you can run flashrom and attach the flasher clip to the GPIO pins on the Ras Pi. The ROM that holds the BIOS and ME firmware is EEPROM. 

Here's an example of external flashing on a ThinkPad T400 (I have a few of these I got for free and have externally flashed most of them to experiment and practice and it works almost exactly as shown on most machines):

https://libreboot.org/docs/install/t400_external.html

mparnelldmp, 2017-10-31:
With my Skylake 2017 Spectre x360, I can flash whatever I want, and it will take long enough to trigger the ME reflash bit set in AMIBCP, but then one of the bios modules triggers a bios recovery when it checks the RSA signature, so my next goal is to try to disable that recovery system entirely since I can just SPI flash a stock bios if I need to...

From there I work on porting coreboot.

Null-Senpai, 2017-10-31:
@mparnelldmp Well since my machine is older I *hope* to not run into that issue.

@matt123b Well I thought I would have to physically remove the chip and put it in a device such as [the thing I linked before](https://www.ebay.com/itm/CH341A-Series-SPI-Flasher-USB-Programmer-24-25-EEPROM-BIOS-Writer-USB-to-TTL/381206013212). The attach-a-clip thing sounds better. 
So all I would do is buy a Raspberry Pi, a [clip such as this](https://www.amazon.com/CPT-063-Test-Clip-SOIC8-Pomona/dp/B00HHH65T4), [some wires](https://www.amazon.com/GenBasic-Female-Solderless-Breadboard-Prototyping/dp/B01L5ULRUA/ref=pd_sim_328_1/134-9110438-2412637) and I can read [my chip](https://user-images.githubusercontent.com/13616397/32244236-d525b150-be6f-11e7-8224-fd7ee8ae8869.jpg)?
The guide you linked mentions a External 3.3V power supply for the chip, is this necessary? Can I not just power on the device in order to supply electricity to the chip or something?
Asking because all those things could be quite expensive for me, especially in contrast to the 3$ CH341A...

mparnelldmp, 2017-10-31:
The ch341a has pins to the side for the wire connections, and it's cheaper, I promise it will work - I own two of them.

If your kernel doesn't have it built in, there's a module, and the latest flashrom syntax:

`flashrom -p ch341a_spi`

Null-Senpai, 2017-10-31:
Oh sweet, that makes things a lot easier (well, cheaper).
Would I still have to get a external 3.3V power supply for the chip though?

mparnelldmp, 2017-10-31:
The USB dongle has both a 3.3 and a 5v pin.

ashgaunt, 2018-06-16:
Hi everyone, sorry if I hikack this thread but I was looking for some advise regarding flashing my HP and you guys seem to be the ones I need to ask. 

I bought an HP Spectre x360 15, but it seems to slow down after using for a few hours. So I monitored it and it seems to be randomlly aplying power throttling (not current or thermal) down to about 16watts. If I fold the screen back in tablet or tent mode it does this immediately and for no reason I can think of. After much research I believe it is just the c state settings int he bios which need to be messed with, but the standard bios has it all hidden.

So I managed to open up the bios in AMIBCP and can see all the settings which I want to make visible, I read that I just need to change them from "default" to "user" and then I saved them over an HP usb flash stick which I made using a virgin downloaded bios flashing app from hp. 

But when I try to flash it I get the signature error thing. Do I need to modify the sig files?
Also why are there two bios files (083BA.bin and 083BB.bin) they apper the same?
Also If I do get it to flash, what are my chances of bricking the laptop? I mean would it still be always possible to reflash back to the original bios with USB key? Would removing the bios battery solve any problems I might have made for myself when I start playing with the setting? (Granted that I dont burn it out some how) Or would I need to send the motherboard off to get the chip reflashed?

I'm quite new at this stuff but thi laptop gets so slow when I am using it hard and I am convinced it does not need to power throttle.