corna/me_cleaner #80: [suggestion] ME disablement via removing VSCC values

skochinsky, 2017-11-02:
VSCC table is present in the descriptor region and tells the ME (and PCH?) about _Vendor Specific Component Capabilities_ of the flash chips supported by the system.

A birdie told me that a _missing VSCC value_ will place the ME into recovery mode since it won't be able to write to flash without knowing if the connected part is supported.

It seems the table is located at VTBA (whose value and size are encoded in the FLUMAP1).

Clearing this table may be another "soft disable" option in addition to AltMeDisable.

platomav, 2017-11-03:
Also SMBus can be found at the FD. Various other settings too if someone wants to go the extra mile. I verified VSCC removal [here](https://www.win-raid.com/t2443f39-Remove-ME-Ignition-firmware-completely-amp-Panther-Point-PCH-observations.html). If tampered with, it should result in a "VSCC_FAILURE" error at PhaseStatus of FWStatusRegister1 as seen [here](https://www.win-raid.com/t2365f39-MSI-GP-PE-Leopard-ME-troubles.html).

raz123, 2017-11-08:
Good catch. More info:

> VSCC Table 
> This section is used to store information to setup flash access for ME. This does not 
have any effect on the usage of the Flash programming Tool (FPT). If the 
information in this section is incorrect, the Intel® ME Firmware may not 
communicate with the flash device. This information provided is dependent on the 
flash device used on the system. More detail please refer to PCH SPI programming 
guide Section 6.4

Source: System Tools User Guide for 
Intel® Management Engine 
Firmware 6.0 (Corporate) -- http://composter.com.ua/documents/System_Tools_User_Guide_for_Intel_Management_Engine.pdf

corna, 2017-11-09:
@skochinsky Doesn't it also affect the normal operation of the BIOS?

skochinsky, 2017-11-10:
> @skochinsky Doesn't it also affect the normal operation of the BIOS?
> 
AFAIK BIOS has its own tables for talking to flash so it should continue working.