corna/me_cleaner #81: ME_Cleaning at FOSDEM 2018

Wiggum127, 2017-11-06:
Dears,

Would anyone be interested in doing a ME_Cleaning of systems during next year's FOSDEM ? All information is available here: https://fosdem.org/2018/

Booths are available and can be used to spread the word, give demonstrations or even get working on systems. Or better still, why not do a lightning talk on the subject or try to enter a devroom ?

What do you guys think ?

matt123b, 2017-11-06:
We could compile a list of machines that work well with ME Cleaner and have people that do go there give a demo. I'd say that a majority of the tech savvy people I know have no idea what the Intel ME does or that it even exists, so informing the audience on it will be important. Heck, I didn't even know about this thing before almost 2 years ago when I discovered Libreboot. The one thing that we really need is detailed documentation on the ME firmware that's easy to understand for the average computer geek. It's incredibly complex and needs proper context. 

I think that a somewhat condensed version of this presentation from Igor Skochinsky would be good:

https://github.com/skochinsky/papers/blob/master/2014-06%20%5BRecon%5D%20Intel%20ME%20Secrets.pdf

citypw, 2017-11-07:
more ME info:
https://github.com/hardenedlinux/firmware-anatomy/blob/master/hack_ME/me_info.md