corna/me_cleaner #82: Not working with Gigabyte GA-Z170X-UD3

syboxez, 2017-11-07:
Made a comment on another related issue, but thought I would make a separate issue since it is a different motherboard.

After POST when flashing using adolfintel's modified Efiflash that doesn't check the checksum (only HAP bit set and nothing else), it shutdown and rebooted every time I tried to load any operating system (either UEFI or BIOS). Was able to reflash the original using the built in utility.

When using me_cleaner without the HAP bit however, it did not boot at all, and it simply flashed from the backup BIOS.

syboxez, 2017-11-07:
Forgot to mention I'm using master branch 312ef02.

corna, 2017-11-23:
> When using me_cleaner without the HAP bit however, it did not boot at all, and it simply flashed from the backup BIOS.

I suppose that only soft disable is supported on your motherboard then.

It seems to me a problem caused by the BIOS, there's not much we can do unfortunately.

syboxez, 2017-11-29:
Is there some way I can modify the stock UEFI to remove or bypass the check for ME?

corna, 2017-11-29:
> Is there some way I can modify the stock UEFI to remove or bypass the check for ME?

I doubt, unless you have experience on reverse-engineering UEFI firmware

> Also -S does the same thing as -s does (boots successfully, but doesn't boot into any OS), which tells me that the HAP bit is successfully disabling ME and is ignoring the missing parts of ME.

Yes, `-S` and `-s` should lead to the same result (Intel ME stops before trying to load the removed modules and it reports the same status to the BIOS).

Limero, 2017-11-30:
I have the same problem with my Gigabyte GA-Z270N-WIFI.
If I do me_cleaner with no options the computer replaces the BIOS automatically at boot with the dual BIOS. If I do either -s or -S the computer keeps rebooting until I reflash stock firmware.

I used the modified flasher from https://www.win-raid.com/t1361f16-Flashing-Gigabyte-while-avoiding-quot-Invalid-BIOS-image-quot.html

syboxez, 2017-12-01:
> I doubt, unless you have experience on reverse-engineering UEFI firmware

At that point, I'd just port coreboot to my board. I already know a bit of x86 ASM, so maybe I could actually do that. I'm just afraid of bricking since I don't have a hardware flasher hooked up (would a Pi Zero work as one?).

corna, 2017-12-09:
> If I do me_cleaner with no options the computer replaces the BIOS automatically at boot with the dual BIOS. If I do either -s or -S the computer keeps rebooting until I reflash stock firmware.

It seems that the BIOS can successfully initialize the hardware without options. You can try to flash both BIOS chips with a modified image, so that the BIOS can't restore the original ME image on boot. Keep an external programmer with you during the process.

> At that point, I'd just port coreboot to my board. I already know a bit of x86 ASM, so maybe I could actually do that. I'm just afraid of bricking since I don't have a hardware flasher hooked up (would a Pi Zero work as one?).

Porting coreboot ranges from super easy (like in the case of Sandy/Ivy bridge desktops) to super hard (Skylake/Kaby Lake laptops). You'll brick it at least once without any doubt, you absolutely need a hardware flasher: the Pi Zero is good for that.

seropp, 2017-12-29:
Commenting just to say Gigabyte GA-B250M-D3H fails in the same way.