corna/me_cleaner #83: Reducing risk of brick.

IwHy4XnXCM, 2017-11-07:
Hi there. I'm a developer but don't have much clue about how this works or how I can avoid bricking my system. I consider myself an end user of this software in that regard.

I have ran the `me_cleaner` check against the BIOS firmware and all looks good.
```
Full image detected
The ME/TXE region goes from 0x1000 to 0x200000
Found FPT header at 0x1010
Found 19 partition(s)
Found FTPR header: FTPR partition spans from 0x48000 to 0xd0000
ME/TXE firmware version 9.0.30.1482
Checking the FTPR RSA signature... VALID
```

I have also ran the injector/remover and all looks good.
```
Full image detected
The ME/TXE region goes from 0x1000 to 0x200000
Found FPT header at 0x1010
Found 19 partition(s)
Found FTPR header: FTPR partition spans from 0x48000 to 0xd0000
ME/TXE firmware version 9.0.30.1482
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x0a)...
Reading FTPR modules list...
 UPDATE           (LZMA   , 0x0a332c - 0x0a3461): removed
 ROMP             (Huffman, fragmented data    ): NOT removed, essential
 BUP              (Huffman, fragmented data    ): NOT removed, essential
 KERNEL           (Huffman, fragmented data    ): removed
 POLICY           (Huffman, fragmented data    ): removed
 HOSTCOMM         (LZMA   , 0x0a3461 - 0x0ab794): removed
 TDT              (LZMA   , 0x0ab794 - 0x0b0b4f): removed
The ME minimum size should be 393216 bytes (0x60000 bytes)
The ME region can be reduced up to:
 00001000:00060fff me
Checking the FTPR RSA signature... VALID
Done! Good luck!
```

I noticed that the documentation mentions external flashing - are there any guides on how to do this?

Just for the sake of documentation, I am using a Z87-G45 motherboard with an Intel i5-4570. Using MSI's latset chipset drivers. 

Thanks.

sakaki-, 2017-11-09:
It doesn't contain a video, but my guide [here](https://wiki.gentoo.org/wiki/Sakaki%27s_EFI_Install_Guide/Disabling_the_Intel_Management_Engine) is based on the RPi3 and has a number of photos, pinouts etc. that you may find useful.

matt123b, 2017-11-09:
External flashing isn't that hard at all as long as you have a decent idea of how it works and what tools to use (this means you should understand what various tools and commands do before using them, don't just copy and paste). When flashing externally, be sure to dump the original ROM not just once, but TWICE and check both images to make sure that they're exactly the same. If they aren't, you have a problem. If both images are the same, you can use either one as a backup if the ME Cleaner image doesn't work. You (usually) won't brick your hardware permanently unless you're heavy handed and physically break something off the board and can't solder it back. I've personally never bricked any machine when flashing the BIOS ROM. Worst case scenario you'll have to restore to the original image.

closed 2017-11-13 by IwHy4XnXCM