corna/me_cleaner #84: What's the me_cleaner support status for SPS?

citypw, 2017-11-09:
Hi,

I tested Intel(R) Xeon(R) CPU E3-1230 v5( Skylake) with ASUS P10S-M WS accidentally cu'z I thought it was using MEv11 instead of SPSv4.0. So I want to ask a few questions before further test on more machines. Has anyone tested it on SPSv4.0? What's the differences between SPS and regular ME? Is it run MINIX-based OS also?

Thanks,

skochinsky, 2017-11-09:
SPS 4.0 is somewhat close to ME11 on the low level (x86 core, Minix-based, $CPD code section layout). It does not include AMT functionality since servers tend to have a dedicated BMC for remote management. Instead it may include Node Manager or Datacenter management functionality (sensor readings/fan control etc.).

corna, 2017-11-09:
The most recent test on SPS is [this one](https://github.com/corna/me_cleaner/issues/71), however it is on SPSv3, so it could be a completely different situation.

citypw, 2017-11-11:
@skochinsky Is BMC build based on SPS or it's running on another independent chip? It's really good to know that SPSv4 is close to MEv11 maybe cu'z both of them running on x86. Thanks for pointing out.

@corna Great, there are other test cases for removing code modules from SPS. Thanks.

skochinsky, 2017-11-11:
BMCs tend to use ARM chips but this depends on board manufacturer. They're often Linux-based.

citypw, 2017-11-12:
Thanks. That's the info I need.

closed 2017-11-12 by citypw

Yannik, 2018-02-25:
@citypw I successfully used me_cleaner on SPS4 (see https://github.com/corna/me_cleaner/issues/3#issuecomment-368335041)

citypw, 2018-02-26:
@Yannik Awesome, I only tested [one machine w/ SPSv4.0](https://github.com/hardenedlinux/hardenedlinux_profiles/tree/master/coreboot) resulted in delayed booting if the SPS was soft disable w/o code module removal. We may need more SPS machines to test. Thanks for the info again!

Yannik, 2018-02-26:
@citypw Yeah, soft disable only didn't work for me either. I think module removal is the way to go here (that worked for me w/o issue).