corna/me_cleaner #85: ME cleaner via USB

Bl00dsoul, 2017-11-09:
Is it possible to make ME cleaner work via USB with this new discovery?

https://www.ptsecurity.com/upload/corporate/ww-en/analytics/Where-theres-a-JTAG-theres-a-way.pdf

mparnelldmp, 2017-11-09:
Related: https://twitter.com/h0t_max/status/928269320064450560

matt123b, 2017-11-09:
@mparnelldmp The comments are great.

mparnelldmp, 2017-11-09:
Here's hoping this leads to not just ME disablement but also more control of the CPU itself too...maybe there will be indications of bin sorted CPU's we can enable more cores on and such.

Likewise, with better access I'm sure the skilled reverse engineers in the community will find any actual back doors in their research.

mparnelldmp, 2017-11-09:
Do we need to crowdfund one of these for Corna and the coreboot guys?

https://www.ebay.com/itm/Intel-ITP-XDP-3BR-JTAG-Debugger/182860599392?hash=item2a93575060:g:SCIAAOSwUoNZ8-iJ

matt123b, 2017-11-13:
@mparnelldmp It will be great to have our suspicions confirmed about the backdoors. People have been trying to warn us for almost a decade about this and no one seems to be listening, save for a few hobbyists and security researchers. What I a really want to know is whether or not US three letter agencies have access or not. Obviously the HAP bit was discovered and so they've been involved to that extent but what I want to know is if they're using the Intel Management Engine as a malicious backdoor to target home users. I know that they love "dragnet" style surveillance.

mparnelldmp, 2017-11-13:
Even if not used to target home users, the non-US Intel chips could potentially have a different build of ME for such things...something worth research there.

I did find this script via googledork on NSA.gov where they're talking about the HAP program, it's pretty telling.

In case you're allergic to .gov URLS: https://archive.fo/aq1VZ
Original: https://www.nsa.gov/resources/everyone/digital-media-center/video-audio/information-assurance/assets/files/orlando-2010-transcript.pdf

More Dorked Files:
https://encrypted.google.com/search?hl=en&q=high%20assurance%20computing%20hap%20pdf%20site%3Agov

corna, 2017-11-23:
@Bl00dsoul answering the original question: maybe in the future, not yet unfortunately. The folks at Positive Technologies are doing a coordinate disclosure with Intel and I think we'll have useful informations about this topic during their [BHEU presentation](https://www.blackhat.com/eu-17/briefings/schedule/#how-to-hack-a-turned-off-computer-or-running-unsigned-code-in-intel-management-engine-8668).

closed 2017-11-23 by corna