corna/me_cleaner #86: MSI JSPI1 and JME1

IwHy4XnXCM, 2017-11-09:
Hi there.

Just taking a look at my motherboard, and I've noticed JSPI1 and JME1 pin headers. According to what people are saying online, the JSPI1 header is a convenient interface for flashing. As for JME1 though, I am confused whether it is related to Intel ME or not.

Could anybody help me find the [pinout for the header](https://i.imgur.com/CsRsChF.jpg)? It doesn't match up with the [current pinout documentation I'm looking at](https://www.flashrom.org/MSI_JSPI1).

Thanks.

skochinsky, 2017-11-10:
JME could be for recovey mode, you could try putting a jumper on it and see if you get different messages on boot and/or in intelmetool output.
JSPI1 pinout looks matching to me with the exception of two extra pins. You can probably confirm it with a multimeter and flash chip's datasheet.

corna, 2017-11-13:
Is it an MSI? It seems identical to [my JSPI1](https://github.com/corna/me_cleaner/wiki/me_cleaner-on-a-MSI-H110M-ECO#step-1-locate-the-spi-flash-chip).

As suggested by Skochinsky, the JME could be the recovery jumper, which could also allow you to disable the descriptor flash restrictions.

IwHy4XnXCM, 2017-11-13:
@corna That's the one! Same pinout. Annoyingly my jumper cables are too large to all fit onto the motherboard's header, but at least I know it's possible now. Could make the wiki section for "JSPI1" rather than mobo specific?

corna, 2017-11-23:
It's probably a proprietary MSI connector but I think a separate wiki page is not needed, I'll just add a note stating that it can be found also on other MSI boards.

closed 2017-11-23 by corna