corna/me_cleaner #87: Lenovo L512 w. phoenix Bios

arjunae, 2017-11-10:
Dear Community,

currently me_cleaner wont parse bios backups from that Machine.
Out of Curiosity, i tried a bunch of bios related tools on the rom and besides one, all failed.
The one who was actually able to parse, was this one: [PhoenixTools](https://www.bios-mods.com/tools/index.php?dir=Andy+P+%28MDL%29+Phoenix-Insyde-EFI+SLIC+Tool%2F) (applies predefined patches to Phoenix based roms) While dumping,  it simply claims that the roms been using the "new" Format.
As this may be of some sort of compatibility interest, for anyone havin those Machines, i attached the Dump here.
[LENOVO-81ET63WW-Dump.zip](https://github.com/corna/me_cleaner/files/1461134/LENOVO-81ET63WW-Dump.zip)

regards, Thorsten

skochinsky, 2017-11-10:
Did you do the dump from the OS? It does not include descriptor and ME regions, probably those are inaccessible to the CPU and you need to use an external flasher.

arjunae, 2017-11-11:
Well i do think so. (posted aboves update before reading, so i will try again with that backup)
Some searching in the net shows, that a similar Model to the L512, a T420 has to be flashed using the way you suggested. https://www.coreboot.org/Board:lenovo/t420