corna/me_cleaner #89: Help 'internal' flash using Ubuntu (Asus Z170 Pro Gaming.)

older-pack, 2017-11-10:
Sorry if this is the wrong place for this help request.

I have an Asus Z170 Pro Gaming motherboard.
Intel i5 6600K CPU
I've seen a couple of reports that the motherboard was flashed successfully (external flash).

I've successfully disabled Intel ME on the BIOS.
But I don't have external flashing tools so that's out.

I'm on Ubuntu I have no access to Windows.
So I can't use the Windows "Aptio AFU Win Flasher" utility as in [this example](https://github.com/corna/me_cleaner/issues/62).

So that leaves me with FreeDOS/MS-DOS flashing using AFUDOS utility.

I've tried a various live CD's that have FreeDOS on them (incl Asus's own CD)
But every time the flash fails with the same error
`22 - Error: Problem alloating memory`

I've seen a couple of reports that the error is caused by FreeDOS and an MS-DOS live disc overcomes the issue.
How do I create a bootable MS-DOS USB disk on Linux?

I've found a few examples for Windows.
I've found [one on Ubuntu](https://ubuntuforums.org/showthread.php?t=817897) (- but it doesn't work)

Any ideas? Thanks

mooms06, 2017-11-11:
On Linux you should be able to use Flashrom. I don't use Linux, so I can't help here.

You should really buy an external SPI programmer though, they are [very cheap](https://www.ebay.com/itm/272575052728), and in case something goes wrong, you can easily flash the backup. [here is a guide](https://www.win-raid.com/t796f16-Guide-Using-CH-A-based-programmer-to-flash-SPI-EEPROM-5.html#msg41915)
If you don't have another machine, then you should also buy [a spare chip on eBay](https://www.ebay.com/itm/401144672038) (this is the one you need for Z170 Pro Gaming, I own the same board).

If you really want to try internal flash, [here is a guide](https://www.win-raid.com/t455f16-Guide-How-to-flash-a-modded-ASUS-ASRock-Gigabyte-AMI-UEFI-BIOS.html).

older-pack, 2017-11-11:
Thank you very much, mooms06.
Maybe you can tell, but I'm in ever-so-slightly over my head!
Those guides you posted are brilliant. Thank you.

Any ideas why external is so preferable to internal flashing?

mooms06, 2017-11-11:
>Any ideas why external is so preferable to internal flashing?

Most of the times you can't flash ( at least easily) the whole chip, because some regions are read/write locked.
Also, if you don't have a backup solution, you will end with a bricked mobo if something goes bad.

Since you don't seem to have a spare machine, buy the SPI programmer and the spare chip I've linked, this way you are 100% fool-proof.

I also advise to use the `-S` flag in me_cleaner, if you don't use it you will have a 15sec delay before POST.

older-pack, 2017-11-11:
> because some regions are read/write locked.

Wow, I didn't realize that.
And I didn't realize the 'SPI programmer' tool was so cheap.

> I also advise to use the -S flag

Now I know who you are!
I had spotted your [comment](https://github.com/corna/me_cleaner/issues/3#issuecomment-342210362) and `-S` is what I had used thanks to you :)
 
Thanks again for all your help.

mooms06, 2017-11-11:
You're welcome, and happy tinkering, you will see it's fun !

closed 2017-11-18 by older-pack