corna/me_cleaner #90: [Request/Question] me_cleaner working for Broadwell-DE SoC (Xeon D-1500) ?

ragingturd, 2017-11-11:
Since the SoC is in many Supermicro Boards with long lifetime, it would be great, to remove Intel ME from all those machines.

But after reading everything I found, I am still not sure if me_cleaner can work on Boards with SoCs or not. And if it does: Are there any successes on the Supermicro X10SDV Series?

ragingturd, 2017-11-18:
Well, I tried to flash it anyway... Is this a success?? I do not think so...

./intelmetool -s
MEI was hidden on PCI, now unlocked
MEI found: [8086:8c3a] 8 Series/C220 Series Chipset Family MEI Controller #1

ME Status   : 0xf0382
ME Status 2 : 0x90403008

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : YES
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Recovery
ME: Current Operation State : Bring up
ME: Current Operation Mode  : (null)
ME: Error Code              : No Error
ME: Progress Phase          : Moff->Mx wake after an error
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : Unknown 0x40

ME: Extend Register not valid

ME: timeout waiting for data: expected 8, available 6
ME: GET FW VERSION message failed
ME: timeout waiting for data: expected 5, available 0
ME: GET FWCAPS message failed
Re-hiding MEI device...done

sakaki-, 2017-11-18:
This looks like a success to me, see e.g. the partial list of telltale output from `intelmetool` [here](https://wiki.gentoo.org/wiki/Sakaki%27s_EFI_Install_Guide/Disabling_the_Intel_Management_Engine#imt_check), and #71 (also an SPS board).
The ME is still visible at a low level over PCI, but is no longer functional.

ragingturd, 2017-11-18:
I looked at the differences in BIOS and this may also hint to a success:

[image]

[image]

corna, 2017-11-23:
> But I didn't flash the final image, because I'm scared.

Perfectly reasonable ;)

> Well, I tried to flash it anyway... Is this a success??

You took a risk but it seems that it paid off. Your situation is different from the usual ones, as you don't have ME/TXE but you have SPS.

The indicators of success are:

> Found 6 partition(s)

Even if it printed just one module, 5 partitions have been removed.

>  ME: Current Operation State : Bring up

SPS is locked in the `Bring up` phase.

>  ME: GET FW VERSION message failed

Self-explanatory

> Operational firmware version: 0.0.0.0

It is common for BIOS to report 0.0.0.0 as version when ME/TXE/SPS is disabled.

Well done!!

closed 2017-11-23 by corna

dongcarl, 2018-02-07:
@corna What flags did you provide to me_cleaner?

ragingturd, 2018-02-07:
@dongcarl I followed the "Internal flashing with OEM firmware" guide:

 $ python me_cleaner.py -S -O modified_firmware.bin original_firmware.bin

Nothing fancy there. Plus it still works today. But BIOS Version 1.3 for all X10SDV Boards from Supermicro is on its way (including Meltdown fixes), and I do NOT know, if it will work on those releases. I hope someone else will play guinea pig this time.

dongcarl, 2018-02-07:
@ragingturd that's exciting because I have the exact same board! Questions if you have the time:
1. Which BIOS version are you on right now?
2. Which method did you use to flash the BIOS? (Recovery or a DOS drive)
3. How stable is the machine with ME removed?

ragingturd, 2018-02-07:
@dongcarl 

1. Which BIOS version are you on right now?
BIOS Version: 1.2c
BIOS Build Time: 09/19/2017
As I meantioned above, Supermicro will release 1.3 at some point, due to CPU fixes (and maybe other stuff, I don't know of). Source: https://tinkertry.com/meltdown-and-spectre-info

2. Which method did you use to flash the BIOS? (Recovery or a DOS drive)
I formatted an USB Stick to be used as DOS Boot device. Was a bit of PITA, I have to say, but it worked. You also may be able to flash it via IPMI, but then you would have to buy an Out of Band Management license and noone of us has proofed this to be a working method.

3. How stable is the machine with ME removed?
Since I didn't measure performance or bootup time, I can't provide you with any information on that topic. But I am using this board as my NAS at home and it does the job in the same manner, as it did before. No complaints there.

In case you are scared: The hell, I was scared as well, but in the end it worked. But still: It may fail in your case for whatever reason. You can only find it out by yourself. If you are successful, the 3rd person won't have to be that scared. :)