corna/me_cleaner #92: Supermicro X10DAL-i, is this considered a success?

Saren-Arterius, 2017-11-12:
```
MEI not hidden on PCI, checking if visible
MEI found: [8086:8d3a] C610/X99 series chipset MEI Controller #1

ME Status   : 0xf0382
ME Status 2 : 0x90409008

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : YES
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Recovery
ME: Current Operation State : Bring up
ME: Current Operation Mode  : M0 without UMA but with error
ME: Error Code              : No Error
ME: Progress Phase          : Uncategorized Failure
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : Unknown 0x40

ME: Extend Register not valid

ME: timeout waiting for data: expected 8, available 6
ME: GET FW VERSION message failed
ME: timeout waiting for data: expected 5, available 0
ME: GET FWCAPS message failed
```

Saren-Arterius, 2017-11-12:
I found `# intelmetool -s`'s  output identical to https://github.com/corna/me_cleaner/issues/71 (without `-s` section). So I think it's a success, waiting for confirmation.

dp-alvarez, 2017-11-21:
@Saren-Arterius Can't help you verify ME is gone, but I'd like some more info on how you flashed your motherboard.
Did you use the AltMeDisable option, did you remove the modules or both? How did you flash it, external flasher or just passed the BIOS image to me_cleaner and used Supermicro's bios flasher?
Planning on cleaning the me from my x11ssm-f, so any info would be very helpful.

sakaki-, 2017-11-21:
@Saren-Arterius, this looks like a success to me, see e.g. the partial list of telltale output from `intelmetool` [here](https://wiki.gentoo.org/wiki/Sakaki%27s_EFI_Install_Guide/Disabling_the_Intel_Management_Engine#imt_check). (Also #71 you mentioned, that was one of my boxes.)

Saren-Arterius, 2017-11-22:
@dp-alvarez I didn't use AltMeDisable. I just ran `python3 me_cleaner.py -O mod.bin X10DAL6.910` than flashed mod.bin using standard procedure provided by Supermicro which is included in BIOS download's zip file. 

@sakaki- Thank you for identifying.

corna, 2017-11-24:
@Saren-Arterius I agree with @sakaki-, it seems a success to me:

> ME: Current Operation State : Bring up
> [...]
> ME: Progress Phase          : Uncategorized Failure
> [...]
> ME: Extend Register not valid

locked in bring up

closed 2017-11-24 by corna