corna/me_cleaner #94: Did I just kill my system?

IwHy4XnXCM, 2017-11-13:
I realise this is not a tech support forum but....

Just tried to hook up my RPi to the JSPI1 header, flashrom was never able to recognise the EEPROM. I reconnected and booted my PC and the monitor isn’t turning on. 

I probably fucked up right? I made sure that the GPIO pins were correctly wired and all.

IwHy4XnXCM, 2017-11-13:
SORRY, my bad, connector stopped working. Unrelated. On the other hand, if anybody could help me diagnose why the EEPROM wasn’t discovered that would be cool. Thanks and sorry again 😂

corna, 2017-11-13:
Happens ;)
Do you have a multimeter?

IwHy4XnXCM, 2017-11-13:
@corna I do, but in honesty I was using some dodgy jumper cables. That's probably where the issue was. I will invest in a SOIC clip and hopefully all will be well. Thank you 👍 

You can delete this issue if you wish, I will close it now.

closed 2017-11-13 by IwHy4XnXCM

corna, 2017-11-13:
Also [these ones](https://www.aliexpress.com/item/12Pcs-Mini-Grabber-6-Color-Test-Probe-Hook-For-SMD-IC-Test-Cilps-SMD-IC-Hook/32684876176.html?ws_ab_test=searchweb0_0,searchweb201602_0_10073_10152_10151_10536_10557_10537_10539_10055_10154_10177_10056_10155_10059_10312_10314_10534_10313_10533_10060_10084_100031_10083_10184_10107_10304_10307_10341_10065_10142_10340_10068_10343_10541_10342_10562_10345_10103_10102_10344_10302-10102,searchweb201603_0,ppcSwitch_0&algo_pvid=d2a452cc-a81a-4f65-a8ba-2ddf5e5157e6&algo_expid=d2a452cc-a81a-4f65-a8ba-2ddf5e5157e6-1) are a good choice, cheap and reliable.