corna/me_cleaner #96: "Unknown image" for Acer Aspire VN7-571G

cron0mat, 2017-11-16:
Hi,

**ifdtool -d HDS114.fd** 

provides good result but 

**./me_cleaner.py -c HDS114.fd**

returns just **"Unknown image"**

My notebook info 

```
# dmidecode 3.1
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.
27 structures occupying 1410 bytes.
Table at 0x9CEBB000.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: Insyde Corp.
	Version: V1.08
	Release Date: 09/12/2014
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 4096 kB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.8
	Firmware Revision: 1.12

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Acer
	Product Name: Aspire VN7-571G
	Version: V1.08
	Serial Number: NXMQKEG0224430D72C6600
	UUID: FBD5FB7D-A698-0447-BBBF-EC046DC7693B
	Wake-up Type: Power Switch
	SKU Number: Aspire VN7-571G_091B_1.08
	Family: Sharkbay System

Handle 0x0002, DMI type 2, 16 bytes
Base Board Information
	Manufacturer: Acer
	Product Name: Aspire VN7-571G
	Version: V1.08
	Serial Number: NBMQK110014430068A6600
	Asset Tag: No Asset Tag
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Type2 - Board Chassis Location
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

```

corna, 2017-11-24:
Can you post the output of `ifdtool -d HDS114.fd`?

cron0mat, 2017-12-01:
here is the output of ifdtool

[ifdtool.txt](https://github.com/corna/me_cleaner/files/1520684/ifdtool.txt)

corna, 2017-12-09:
It seems that HDS114.fd is not a valid BIOS image, see this recently updated [Wiki page](https://github.com/corna/me_cleaner/wiki/External-flashing#dump-of-the-original-firmware). You should try to obtain a valid dump, maybe with an external programmer.