corna/me_cleaner #97: Is me_cleaner supposed to work on Thinkpad X1 Carbon latest generation (gen 5th)?

2Belette, 2017-11-18:
Everything in the subject :)

pedrib, 2017-11-21:
I'm interested in finding this out too, as I want to get one soon.

sakaki-, 2017-11-21:
According to [this post](https://www.reddit.com/r/openbsd/comments/6xt7bc/openbsd_on_the_lenovo_thinkpad_x1_carbon_5th_gen/dmmv652/), the current generation of X1 has Boot Guard enabled; and [per Nicola Corna:
](https://github.com/corna/me_cleaner/wiki/Intel-Boot-Guard)

> Intel Boot Guard is a technology introduced by Intel in the 4th Intel Core generation (Haswell) to verify the boot process. This is accomplished by flashing the public key of the BIOS signature into the field programmable fuses (FPFs), a one-time programmable memory inside Intel ME, during the manufacturing process; in this way the CPU contains the public key of the BIOS and it can verify its correct signature during the boot. Obviously, once enabled by the manufacturer, Intel Boot Guard can't be disabled anymore.
> Unfortunately for us Intel Boot Guard is not compatible with me_cleaner as the machine will not power on if Intel ME has been disabled, **even if the BIOS hasn't been modified**.

See also issue #6 (wherein it [was confirmed](https://github.com/corna/me_cleaner/issues/6#issuecomment-268788911) that an X1 Carbon gen 2 had shipped with Boot Guard on).

**However**, the possibly good news is that (per [this comment at the end of #6](https://github.com/corna/me_cleaner/issues/6#issuecomment-338989159)) although you cannot purge the ME's firmware modules on a Boot Guard machine, you _may_ be able to use the `-s`/`--soft-disable-only` flag in `me_cleaner` (to set the [HAP bit](https://github.com/corna/me_cleaner/wiki/HAP-AltMeDisable-bit) only), and still have it boot OK.

2Belette, 2017-11-21:
@sakaki- thanks! setting HAP bit seems a good alternative.
I am ready to test on my X1 Gen5th but I am unable to dump my firmware...any idea ?

sakaki-, 2017-11-21:
I don't have an X1 to hand myself, but if you don't mind opening the case then dumping / reflashing via an [external SPI programmer](https://github.com/corna/me_cleaner/wiki/External-flashing) is probably the most reliable method. See e.g. my guide [here](https://wiki.gentoo.org/wiki/Sakaki's_EFI_Install_Guide/Disabling_the_Intel_Management_Engine) (uses an RPi3 as the programmer), or [this guide](https://hardenedlinux.github.io/firmware/2016/11/17/neutralize_ME_firmware_on_sandybridge_and_ivybridge.html) (uses a BeagleBone Black, targetting an X220).

2Belette, 2017-11-21:
thanks for the links, I was ready to try from internal but unfortunately I am not confident to try from outside and even less on this 2K€ machine..as a non expert I have probably chance to brick my system to easily.. :( you think doing it from internal would not work or too complicated? I was just tryin to find a way to dump the UEFI binarie

corna, 2017-11-21:
@sakaki- Unfortunately I haven't found yet the time to update that page. Yes, it seems that our early observations about the interaction between ME and BG were incorrect, and it is indeed possible to disable Intel ME even with BG enabled. However it seems that the required modules and partitions in that case are not just the standard ones, as the PC doesn't turn on without options or with -S. As soon as I have some spare time I'll try to establish which modules are really needed with BG.

@2Belette each method has its own pro and cons. I always prefer the external flashing as, even if something goes wrong, I always have a way to restore the original firmware (and if you brick your laptop with internal flashing, you're forced to unbrick with an external programmer anyways), but it requires a bit of confidence with the hardware.

pedrib, 2017-11-27:
@2Belette I might have found a way. 
Lenovo have released a tool to update Intel ME firmware directly - see https://pcsupport.lenovo.com/jp/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-type-20hr-20hq/downloads/ds120415

I downloaded the tool, and after extracting it it creates two ME images and a bunch of other files in the extraction folder:
ME_11.8_Corporate_C0_LP_Production.bin
ME_11.8_Consumer_C0_LP_Production.bin

I have run me_cleaner through them and the script detected the ME images and cleaned them!
Now I suspect that if you replace the original images with the cleaned ones and then run the executable it will most likely flash the modified images. Do you want to try doing that?

platomav, 2017-11-27:
Nothing to see there. You cannot flash nuked firmware via FWUpdate tool. Both FWUpdate & CSE will immediately reject the firmware.

Also, and this is potentially important, partition IDLM (first appeared at 11.8, only PCH-LP) is probably needed for the INTEL-SA-00086 fix so people should test its removal.

pedrib, 2017-11-27:
@platomav so you think it's not even worth trying?

platomav, 2017-11-27:
FWUpdate tool is used to update the Engine firmware on the field. All OEMs bundle it with new firmware to update end-user systems. They often also bundle MEInfo tool to check the Engine's status post-flash. There is nothing to try or test, everything in that package is normal/expected.

pedrib, 2017-11-27:
@platomav got it. Looks like ISP is the only way to go.

troyready, 2017-12-21:
Curious if anyone can confirm that the 5th gen has a chip that can accessed via a test-clip? I've been looking online and haven't been able to see the 5th gen chip definitely.

I'm made the mistake of getting another ultra-thin Dell notebook that uses a Winbond WSON chip that (as far as I can tell) can't be clipped, so I'm a little gun shy on another thin notebook (tempted instead to just get a T470s instead that has been confirmed working).

villesundell, 2018-01-12:
Any update on this X1 Carbon 5th gen question? Any information would be much appreciated :)

BunnyTheLifeguard, 2019-04-12:
Hey guys, I tried it on a 4th Gen Skylake Carbon using @sakaki- s guide.

When using `me_cleaner --soft-disable original.rom --output modified.rom` the laptop did not boot.
I tried using `me_cleaner -s / --soft-disable original.rom --output modified.rom` but this gives me _me_cleaner: error: unrecognized arguments: original rom_

So then I used just `me_cleaner -s original.rom --output modified.rom` which resulted in a much shorter output (no removing things, only setting HAP bit) and after that laptop boots again, but what confuses me is the `sudo intelmetool --show` output:

`Bad news… blabla`
`RCBA at 0x00000000`
`MEI not hidden on PCI, checking if visible`
**`MEI device not found, huh?`**
`exiting`

Is this the expected behavior?

Thanks in advance.

reasv, 2019-04-19:
@BunnyTheLifeguard To be clear, you were successful in disabling ME on your 4th Gen (2016) Thinkpad X1 Carbon? I have the same one and I'm looking to do this as well but I don't want to brick my laptop.
So just doing me_cleaner -s original.rom --output modified.rom works?

BunnyTheLifeguard, 2019-04-20:
> @BunnyTheLifeguard To be clear, you were successful in disabling ME on your 4th Gen (2016) Thinkpad X1 Carbon? I have the same one and I'm looking to do this as well but I don't want to brick my laptop.
> So just doing me_cleaner -s original.rom --output modified.rom works?

Yup, working fine since reflashing the modified rom :)

reasv, 2019-04-20:
@BunnyTheLifeguard what do you use for flashing? Thanks.

BunnyTheLifeguard, 2019-04-20:
> @BunnyTheLifeguard what do you use for flashing? Thanks.

Hey @reasv , I used a Raspberry Pi and just followed @sakaki- "s guide :)

sakaki-, 2019-04-20:
Hi @reasv, the guide @BunnyTheLifeguard just referred to may be found [here](https://wiki.gentoo.org/wiki/Sakaki's_EFI_Install_Guide/Disabling_the_Intel_Management_Engine).

BunnyTheLifeguard, 2019-04-26:
@sakaki- @reasv yeah, mb, forgot to link it ^^'

pedrib, 2020-05-02:
I think now it's pretty clear that using -s works on the X1 Carbon G5. However I wanted to find a way to soft flash it to avoid having to hard flash it every time a new Lenovo BIOS update comes out (which is quite frequent these days).
Since I followed the [guide here to flash with FPT](https://github.com/mostav02/me_cleaner_wiki/blob/master/Internal-flashing-with-FPT.md) and was able to flash my Dell 7060 Micro successfully, I wanted to see if I could also do it on the X1 G5. 

Booting into Windows, I was able to check that the ME is enabled for writing, but the FD is not. Since Boot Guard is enabled, I'm not going to try to use me_cleaner with -S, as that would probably brick my laptop.

The Dell Micro had a service jumper, so unlocking the FD was easy. The X1 G5 doesn't, so I tried following the [guide by mostav02 here](https://github.com/mostav02/me_cleaner_wiki/blob/master/Internal-flashing-with-FPT.md#bios-options--hidden-options) and [the guide by @platomav here](https://www.win-raid.com/t3553f39-Guide-Unlock-Intel-Flash-Descriptor-Read-Write-Access-Permissions-for-SPI-Servicing.html) so try and set EFI vars that enable ME / FD flash unlocking.

Right off the bat I could tell things were not good - I tried mounting and remounting the EFI partition on Linux and couldn't not write anything. Then I followed what the guides say and booted into Windows again, made a full dump of my BIOS using FPT and loaded it into UEFIExtractor / IRFExtractor (results attached).

I identified two possible vars for modification:
```
GUID: 5432122d-d034-49d2-a6de-65a829eb4c74
Name: "MeSetupStorage"
Attributes:
        Non-Volatile
        Boot Service Access
        Runtime Service Access
Value:
00000000  00 01 01 00 00 00 00 01  00 0d 00                 |...........     |
0x7C1C4 			One Of: ME State, VarStoreInfo (VarOffset/VarName): 0x2, VarStore: 0x1108, QuestionId: 0xAD4, Size: 1, Min: 0x0, Max 0x1, Step: 0x0 {05 91 A1 03 A2 03 D4 0A 08 11 02 00 10 10 00 01 00}
0x7C1D5 				One Of Option: Disabled, Value (8 bit): 0x0 {09 07 04 00 00 00 00}
0x7C1DC 				One Of Option: Enabled, Value (8 bit): 0x1 (default) {09 07 03 00 30 00 01}
```

```
0x7C3FA 	Form: Firmware Update Configuration, FormId: 0x1043 {01 86 43 10 B9 03}
0x7C400 		One Of: Me FW Image Re-Flash, VarStoreInfo (VarOffset/VarName): 0x3, VarStore: 0x4, QuestionId: 0xAE0, Size: 1, Min: 0x0, Max 0x1, Step: 0x0 {05 91 BB 03 BC 03 E0 0A 04 00 03 00 10 10 00 01 00}
0x7C411 			One Of Option: Disabled, Value (8 bit): 0x0 (default) {09 07 04 00 30 00 00}
0x7C418 			One Of Option: Enabled, Value (8 bit): 0x1 {09 07 03 00 00 00 01}
0x7C41F 		End One Of {29 02}
```

I then modded and rebuilt the patched grub in the guides above to try to modify these two vars, but unfortunately no success. When I boot into the EFI shell and run setup_var on any of those two, I get the message:
```
error: can't set variable using efi (error: 0x00000000000000000008)
```
I'm pretty sure the patches I made to grub are correct, as it identifies the VarStore size, GUID and name correctly in both cases, so I'm guessing this means that Lenovo locks down these variables in NVRAM?

closed 2024-07-25 by 2Belette