corna/me_cleaner #98: [Tutorial] Flash Asus Z170 Pro Gaming using Ubuntu

older-pack, 2017-11-18:
**Warning. Follow this at your own risk. You could brick your computer.
Read the me_cleaner Wiki pages first.**

I'm new to all this and it took quite a while to gather all this information.
So to save anyone else the trouble here's what I learned.

You'll need access to a second computer to flash the BIOS chip of your main computer.
(Or you could buy a second BIOS chip - they are cheap - flash it and then swap them.)

You can flash the BIOS chip from a 64 bit computer using a 32 bit computer.

The Asus Z170 Pro Gaming is a great motherboard for flashing the BIOS as the chip is easily removed.
It just plugs in and out of a socket. That makes flashing easier.
The location of the chip is shown on page 1-2 of the Asus User Guide. Its located below the Z170 chip.
[image]

Its best to flash the BIOS chip 'externally' as some regions of the chip might be locked or inaccessible if you try flashing 'internally'.
The tool to flash the chip externally (the USB 'Programmer') is very inexpensive. Mine was ~$5.
Search ebay for "CH341A USB bios board programmer 24 25 series"
(I tried the internal method using various versions of FreeDOS and various DOS flashing tools and ALL were a waste of time.) 

Make note of any changes you have made to your BIOS setup as they will be lost after flashing.

You can run a couple of commands before and after the flash that will confirm Intel ME is disabled.
Before:
```
$ ls /dev | grep mei
mei0
	
$ lspci | grep -i communi
00:16.0 Communication controller: Intel Corporation Sunrise Point-H CSME HECI #1

$ cat /proc/bus/pci/devices | grep a13a
(It returned a couple lines of numbers and spaces and 'mei' was at the end.)
```

They all confirm that Intel ME is on my system.

## Extract the BIOS .ROM from the Ausus .CAP with UEFITool
[This example](https://github.com/corna/me_cleaner/issues/62) explains that you need [UEFITool](https://github.com/LongSoft/UEFITool) to extract the BIOS rom from the .CAP file that Asus uses. me-cleaner can't work with .CAP files. There are Windows and OS X apps ready to use but on Linux you have to compile your own version. I don't know how to do that but I had access to a mac so I used the OS X version. I used [ver 0.21.5](https://github.com/LongSoft/UEFITool/releases/download/0.21.5/UEFITool_0.21.5_osx.zip) as in the example.

And I used it on Asus' current BIOS [ver 3501](http://dlcdnet.asus.com/pub/ASUS/mb/LGA1151/Z170-PRO-GAMING/Z170-PRO-GAMING-ASUS-3501.zip).

Open the Asus .CAP file with UEFITool and highlight the top line "AMI Aptio Capsule". Right click on it and select "Extract body". Save it with .rom extension.

## Disable Intel ME
This may be obvious to some but it took me a while to find it.
There is a built in help in the me_cleaner script:
`$ python me_cleaner.py --help`

User [mooms06 says](https://github.com/corna/me_cleaner/issues/3#issuecomment-342210362) the -S option (--soft-disable) is best for the Pro Gaming as it prevents a 15 sec delay on startup that -s (--soft-disable-only) causes.
To disabled Intel ME I used:
`$ python me_cleaner.py -O modified.rom -S Z170-PRO-GAMING-ASUS-3501.rom`

## Remove the BIOS chip
Power off your system fully and unplug it from the mains.
Its a good precaution to discharge and staic from your body by touching something metal.
Locate the BIOS chip and remove it.
Take note of the direction the indent on the chip's top is facing, thats important.
Be **VERY CAREFUL** removing it. I jerked removing mine and bent the pins!
[image]

Thankfully none were broken and I was able to straighten them.
I have a sore arm so it was more difficult for me. 
With a bit of care it shouldn't be an issue.

Plug the chip into the USB programmer tool.
Again, take note to the chip orientation.
It should be like this (note the notch on the chip top):
[image]

[image]

## Check the USB programmer is working
On the second computer install flashrom if its not already installed.
Plug in the USB programmer tool.

I loosely followed [this guide's](https://wiki.gentoo.org/wiki/Sakaki%27s_EFI_Install_Guide/Disabling_the_Intel_Management_Engine) flashrom process.

First I read (extracted) the chip's existing BIOS.
```
$ sudo flashrom -p ch341a_spi -r extracted1.rom

flashrom v0.9.9-r1954 on Linux 4.9.0-4-686-pae (i686)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Found Winbond flash chip "W25Q128.V" (16384 kB, SPI) on ch341a_spi.
Reading flash... done.
```

That took 3 mins 20sec.

Read a second copy of the current BIOS

```
$ sudo flashrom -p ch341a_spi -r extracted2.rom

flashrom v0.9.9-r1954 on Linux 4.9.0-4-686-pae (i686)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Found Winbond flash chip "W25Q128.V" (16384 kB, SPI) on ch341a_spi.
Reading flash... done.
```

Now compare the two extract BIOSes with 'diff'

`$ diff extracted1.rom extracted2.rom`

There should be no result if both are the same.
If there is a difference stop and [read this](https://wiki.gentoo.org/wiki/Sakaki%27s_EFI_Install_Guide/Disabling_the_Intel_Management_Engine) and follow his recommendations.

I don't have 'ifdtool' tool to confirm the extracted .roms are BIOSes (and not just junk data). Instead I took a look at extracted1.rom in UEFITool and I could see it was ok. 
That confirms the UBS Programmer and BIOS chip setup is working.

## Flash the BIOS
Here we go:
```
$ sudo flashrom -p ch341a_spi -w 'modified.rom'
	
flashrom v0.9.9-r1954 on Linux 4.9.0-4-686-pae (i686)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Found Winbond flash chip "W25Q128.V" (16384 kB, SPI) on ch341a_spi.
Reading old flash chip contents... done.
Erasing and writing flash chip... Erase/write done.
Verifying flash... VERIFIED.
```

That took about 7mins.

Unplug the USB programmer, remove the BIOS chip and return it carefully to your motherboard.

## Reboot your system
The moment of truth.
Start up your PC and go into the BIOS setup by pressing DEL at boot.

You can confirm Intel ME is gone by going into 'Advanced Mode' and under the 'Main' tab the 'ME Firmware Version' should read 0.0.0.0. 
Before it would have read something like 11.6.11.1092
[image]

Don't forget to re-apply any modifications you had previously made to your BIOS.

Exit the BIOS and continue into Ubuntu.

You can run the same commands you ran before the flash for comparison:
`$ ls /dev | grep mei`
[nothing]
	
`$ lspci | grep -i communi`
[nothing]

`$ cat /proc/bus/pci/devices | grep a13a`
[nothing]

The fact they returned nothing shows Intel ME has been disabled.
I'd run 'intelmetool' but you need to compile it yourself and I'm not able to that.
But the checks we've run are pretty clear that Intel ME is gone.

Finally confirm the computer stays on for more than 30 mins and your are done.
Your computer is Intel ME free.

Thanks to corna and contributors for me_cleaner and thanks to mooms06 for his help. [His answers](https://github.com/corna/me_cleaner/issues/89) might help you further.

And don't for get to donate to corna if you can.

mooms06, 2017-11-19:
Hi, great guide!

A few remarks:

- You don't need to use UEFI Tool to removing the capsule header, in fact you don't need to remove the capsule header since you will be using your own made backup to run me_cleaner. 
It's still a good idea to explain how to use a "vanilla" BIOS file from ASUS though, but they don't contains the UUID (S/N, MAC address...) These can be transferred from a backup to an other image with [FD44Editor](https://github.com/LongSoft/FD44Editor/releases).

- I said the 15 seconds delay was when I didn't used any argument (other than `-O`). I didn't tried `-s `but my bet is it's working as well as `-S`.

- It may be a good idea to update the ME before the cleaning, to avoid any security flaws in the remaining parts. You can find the latest versions [here](https://www.win-raid.com/t596f39-Intel-Management-Engine-Drivers-Firmware-amp-System-Tools.html). I've successfully updated to 11.8.50.3425 (with latest BIOS v3501) before the cleaning.

- Before flashing, you can also update the CPU microcodes with [UBU](https://www.win-raid.com/t154f16-Tool-Guide-News-quot-UEFI-BIOS-Updater-quot-UBU.html), since ASUS didn't bothered to update them with the latest BIOS (they correct the infamous Hyperthreading bug). I've updated them and everything works fine.

- The SPI Programmer with the black PCB [need a hardware fix](https://www.win-raid.com/t796f16-Guide-Using-CH-A-based-programmer-to-flash-SPI-EEPROM-6.html#msg43473) (it output 5V instead of 3.3V on some pins). Agin, modded it myself, it was very easy. The Green one don't have this issue.

older-pack, 2017-11-20:
> You don't need to use UEFI Tool to removing the capsule header...

Great tip. It never crossed my mind that you could bypass UEFITool by using the extracted rom instead.

> but they don't contains the UUID (S/N, MAC address...) These can be transferred from a backup to an other image with FD44Editor.

Why might you want to transfer those to a vanilla BIOS?

> The SPI Programmer with the black PCB need a hardware fix (it output 5V instead of 3.3V on some pins)

Oh wow! I didn't know that. I didn't have any issues though. Luckily DIP8 BIOS chips are apparently ok with it. Phew!

mooms06, 2017-11-20:
>Why might you want to transfer those to a vanilla BIOS?
(Have I just skipped an important step?)

To keep the MAC address, motherboard serial number, and other unique numbers, see the images [in this post](http://forum.hardware.fr/hfr/Hardware/carte-mere/downgrade-cartes-chipsets-sujet_919459_1.htm#t8483831). 

>I didn't have any issues though. Luckily DIP8 BIOS chips (like the Pro Gaming's) are apparently ok with it. Phew!

The same here, but better be on the safe side, the mod is easy to do is you have a soldering iron.

I've made a new discovery: apparently, updating the ME to V11.8 before cleaning caused some aleatory graphic crashes, I've swapped the chip (I have three of them) to a cleaned but not updated ME (v11.6) and it seems the crashes have stopped. Need confirmation though, since the crashes are aleatory.

older-pack, 2017-11-20:
Your right! The LAN mac has been changed! (WiFi mac on separate card is the same)
Oops! I didn't know about that step step.
I didn't see anyone else mention it in the guides I cited above either

I thought that MAC address & S/N etc were hard-coded somewhere else on the motherboard.
I presumed they'd be picked up automatically by the new BIOS just like they are when you update with a regular BIOS.

So then the easiest method is to just use the extracted BIOS rom and that will preserve them.

Its worth pointing out that even with out the MAC address S/N UUIDs etc everything works on Linux! (I guess the main down side would be Windows licenses would be messed up with any changes to them?)

mooms06, 2017-11-20:
When you flash a new BIOS with the tools provided by the vendors, they don't write the full chip, only some area/regions (usually the BIOS region obviously, and sometimes they also update the ME region).

When you do a full write with a "vanilla" image, or when you buy a new chip, they don't have these unique serial number. Usually the machine will still works fine though, even in Windows.

In your case, yes, the easiest method is to just use the backup of the BIOS rom you've made with the SPI programmer. 

You should also run UBU to update the CPU microcodes (use a Windows VM). If you can't, I can do it for you, just upload your ROM image, send me the link, and I will send you back an updated one.
Alternatively, I've made one and sharing it [here (post#6)](https://vip.asus.com/forum/view.aspx?id=20170628023935527&SLanguage=en-us&board_id=1&model=Z170+PRO+GAMING&page=1), you can run me_cleaner on it then transfer your unique info with FD44Editor.

older-pack, 2017-11-20:
I'm pretty sure Linux applies the microcode update on-the-fly at each boot via the kernel.
Checking with `$ dmesg` confirms an update has been applied.
Luckily, I've the i5 66000K CPU which doesn't have hyper threading. So I dodged that Intel mess.

Thanks for the offer though.

mooms06, 2017-11-20:
You're right, if you have a recent kernel the update is applied after the boot, and since you don't have HT, you aren't affected anyway.

corna, 2017-11-25:
Thanks for sharing your experience, I've added your guide in the ["How to apply me_cleaner"](https://github.com/corna/me_cleaner/wiki/How-to-apply-me_cleaner) page.

closed 2017-11-25 by corna

DeathSupportsLife, 2019-04-21:
@mooms06 Do I need to worry about microcode? I'm applying the latest at boot time currently, but I'm about to go through the process of removing Intel ME from my computer so I figure might as well update microcode, if necessary.

mooms06, 2019-04-22:
Yes, it's a good idea.

0181505050, 2019-05-14:
do you have the links for the exact driver you used to get the usb programmer running? and how you installed it? I only moved to linux 3 months ago and im already in the deep end here :) im running zorin which is the latest unbuntu. 
i never even compiled something before. thanks

0181505050, 2019-05-14:
Im at the stage where i have the tool, and i have a spare bios chip. I guess my next move is to download the latest bios cap file from asus which i have. 
MAXIMUS-VIII-HERO-ASUS-3802.CAP
extract the rom. 
run the me cleaner tool on it, 
and write it to the spare chip,
swap them over and cross my fingers?

0181505050, 2019-05-14:
ok i feel a bit stupid. but, for others who are wondering what drivers to install to get the USB flash tool running. i installed flashed using the command.

sudo apt-install flashrom

opening a terminal in the same directory as the modified rom, i executed 

sudo flashrom -p ch341a_spi -w 'modified.rom'

Calibrating delay loop... OK.
Found Winbond flash chip "W25Q128.V" (16384 kB, SPI) on ch341a_spi.
Reading old flash chip contents... done.
Erasing and writing flash chip... Erase/write done.
Verifying flash... VERIFIED.

It appears I have managed to flash the latest rom for my board onto the new chip. the only thing left to do is swap them over. but first i need to make some recordings of the bios settings.
I didn't bother recording the details such as mac address or serial number.

Ill let you know how it goes later.

mooms06, 2019-05-14:
You should not flash a "vanilla" BIOS from ASUS with the programmer, because it lacks the UUID (board s/n, MAC Address, etc..).

The correct way is to dump your BIOS with the programmer, then remove the ME with me_cleaner from this backup, then flash  the modified file with the programmer

0181505050, 2019-05-14:
hi mooms06. Yea, I did actually work. But I was unhappy with the mac address. I tried entering randomly generated value for the UUID, and MAC, and serial and re flashed. However when I got into ubuntu I had no network card at all.

0181505050, 2019-05-14:
is there a way to generate  a fresh UUID and MAC and serial that will keep the network card alive? I know you said I should use the old credentials but I fancy changing it all/.

mooms06, 2019-05-15:
I don't know. [FD44Editor](https://github.com/LongSoft/FD44Editor/releases) can edit/inject these UUID, but it was nor updated since a long time.

0181505050, 2019-05-15:
yea that did actually work. but there must be something about the uuid's and mac's being matched. as when i got in, the network card simply didnt show up. if anyone has any ideas. let me know thanks

weasalNZ, 2020-04-15:
Is there anyway to salvage a bad flash. Downgraded to 1101 and now its cpu led stuck. Hoping I can just get a eeprom chip and flash over with out losing all the settings?

mooms06, 2020-04-15:
Don't you have made a backup before ?
You should be able to flash your EEprom again as long as it's not physically damaged.
You can find EEProms on eBay, example for Z170 (type the ref of your EEprom): https://www.ebay.com/itm/401144672038

weasalNZ, 2020-04-16:
Stupid me has never done one on either of the boards I've flashed. When I used the uefi tool. I "extract as is" not "extract body" and then flashed. Using fpt.exe -bios -f z170~1.rom

weasalNZ, 2020-05-17:
> Don't you have made a backup before ?
> You should be able to flash your EEprom again as long as it's not physically damaged.
> You can find EEProms on eBay, example for Z170 (type the ref of your EEprom): https://www.ebay.com/itm/401144672038

I managed to fix it. I ended up using Linux and the flashed perfectly fine. I had nothing but issues in windows and various machines.

schattenpanzer, 2020-06-11:
Great guide, ASUS Z170-AR confirmed working. No problems whatsoever.

midn1, 2021-03-04:
Does anyone know if this guide should apply to the Z170-A also?

I'm not asking for a guarantee, of course, I just don't know the similarity of the two.

mooms06, 2021-03-04:
Yes, it will works on any Z170 board, as long as the BIOS chip is mounted on a socket (DIP8).

ismay, 2022-01-05:
I've been looking into trying this with my Z170i Pro Gaming MB. I believe the bios chip it uses is a Winbond W25Q128FV. However they seem to be hard to find. Does anyone have a good source for these? Official winbond retailers all seem to have discontinued the chip, and I wouldn't like to try this without a backup chip.

mooms06, 2022-01-05:
They can be found on eBay and Aliexpress

Examples:

https://www.aliexpress.com/item/1005002112323822.html
https://www.ebay.com/itm/233220150344

ismay, 2022-01-12:
> They can be found on eBay and Aliexpress
> 
> Examples:
> 
> https://www.aliexpress.com/item/1005002112323822.html https://www.ebay.com/itm/233220150344

Ah nice, thanks!

Hyzle, 2022-05-29:
can you share z170 pro gaming bios without intel me

mooms06, 2022-05-29:
No because all my backups have the UUID of my card. Note that you can also use [CoffeeTime](https://mega.nz/folder/4oRViKSI#nn-8ZRHBa6_PDwNLzgubNA) to disable the ME, it may be simpler to use if you don't like command line tools like ME_Cleaner, and it can bue used to modify the BIOS to let you use a Coffee Lake CPU (Intel gen 8 and 9) on Z170.

CoffeeTime guides:

Russian: https://forums.overclockers.ru/viewtopic.php?p=15924512
English: https://linustechtips.com/topic/1118475-guide-running-coffeelakerefresh-cpus-on-skykabylake-motherboards/
French: https://forum.hardware.fr/hfr/Hardware/carte-mere/coffeetime-coffee-9700k-sujet_1040725_1.htm

Hyzle, 2022-06-02:
not for coffee lake I need for xeon.

Hyzle, 2022-06-02:
can you record a video for this I dont understand and my english is a little bad

AuroraViola, 2023-07-02:
Thank you so much for this guide because I unbricked my motherboard with this