corna/me_cleaner #99: [Success] AsRock X99 Taichi

Grimeton, 2017-11-19:
Hello.

Aaah the smell of success.

I'm running an AsRock X99 Taichi with the latest Bios and an Xeon 1620v4 and was able to disable the ME via the -S switch. 

I tried the modified Bios image first but the system is not able to initialize the chipset after flashing the modified image. I actually know this because the board has the "Dr. Debug" LEDs that show the status. After turning it on it goes over all kinds of initialization and then hangs at status "67". Regarding to the handbook this means "Chipset initialization error". 

After hanging there for round about ten seconds it then goes on and finally stops at status "79" which refers to the same status as "67". I'm certain those describe different states but the detailed information is not available from the handbook.

Also I noted a few specific things when it comes to the Bios and the tool to flash it.

First a picture when the ME was still enabled:

[image]

And after disabling it via the -S switch:

[image]

It's clearly visible that the version has changed to "N/A". But also note the "hint" written beneath the version number about reflashing when the version should show "N/A". I'm still testing the 31 minutes "barrier"...

I also noted a few things when flashing:

First of all the flash utility comes with some interesting options... 

[image]

Note the /ME and /HOLE switches. 

Also when flashing it's rather slow then gets faster and then slow again. It's slow from 0 to 48%, then fast till 87% and then slow till it hits the end. I wonder if that's somehow important to the process?

Anyway the "-S" switch did the trick. I'd be more happy if removing most of the firmware would work but this is better than nothing.

Thanks a lot!

KR,

G.

corna, 2017-11-24:
So it does work with `-S` (capital) but not without options, doesn't it?

cocafe, 2017-11-29:
@Grimeton Could you checkout the system frequencies after removing Intel ME? Any differences? And do the BCLK tweaks in UEFI still work properly?

Grimeton, 2017-11-29:
@corna Yes, only '-S' works.

@cocafe Anything special to look for or just change the BCLK?

cocafe, 2017-11-29:
@Grimeton Yes, please check out the issues i just posted [https://github.com/corna/me_cleaner/issues/107](url)

I have clock issues on my asrock x99 chip after removing Intel ME functions.

Grimeton, 2018-06-04:
Ok, just a quick heads up:

I just updated the system from BIOS version 1.50 to 1.80 using the "-S" switch again (capital "S").

Worked like a charm. ME is disabled, and I was able to flash the BIOS through the builtin flash tool and didn't have to create a bootdisk anymore.

Thanks again!

KR,

G.