corna/me_cleaner #101: Intel ME CVEs

ArchangeGabriel, 2017-11-20:
Not really an issue, but something I thought should be shared here:
https://security-center.intel.com/advisory.aspx?intelid=INTEL-SA-00086&languageid=en-fr

pedrib, 2017-11-21:
Let's wait until the BIOS manufacturers release their versions... and pray that they didn't cripple me_cleaner.

CuleX, 2017-11-21:
I'm not sure on this, but couldn't the arbitrary code execution be used to set the HAP bit without requiring hardware, effectively disabling the ME's most harmful parts?

On another note, is there some kind of public/semi-public ME research group?

pedrib, 2017-11-21:
It most definitely could, but:
a) this was found by Intel, unlikely that they will release an exploit
b) it won't work in the patched version, which might be the one that breaks me_cleaner

CuleX, 2017-11-21:
a. From the announcement:

> Intel would like to thank Mark Ermolov and Maxim Goryachy from Positive Technologies Research for working collaboratively with Intel on a coordinated disclosure for CVE-2017-5705.

That's the arbitrary code execution.

b. Yes, that's correct. But nobody is keeping people from *not* patching their ME if someone's working on a solution to abuse the arbitrary code execution to set the HAP bit. I believe it would be an endeavor worth undertaking. Obviously, it'd take a coordinated effort unless Positive Technologies Research would feel cooperative on this, but they're likely behind an NDA with Intel wrt disclosure, even if they wanted to help this idea.

pedrib, 2017-11-21:
@CuleX but that is exactly what me_cleaner does. Up until this patch, you can use me_cleaner with two modes, one of them enables the HAP bit. Check the docs. After this patch, who knows.

Grimeton, 2017-11-21:
Question:

If ME is disabled then I don't need to update the BIOS anyway - right?

pedrib, 2017-11-21:
You still need to update your bios. The bios might still have vulnerabilities in other components. ME is only a small part of the firmware, unrelated to the BIOS.

grudnevkv, 2017-11-21:
Expect to see PoC or some exploits at BH2017
https://www.blackhat.com/eu-17/briefings/schedule/#how-to-hack-a-turned-off-computer-or-running-unsigned-code-in-intel-management-engine-8668

Grimeton, 2017-11-21:
@pedrib Yeah I need to update the bios if there is a new version of the Bios or a vulnerability is found. But as the ME part is not used, changes to the ME part don't affect me anyway so I don't have to update the Bios do I?

pedrib, 2017-11-22:
I still don't understand your question. Why would you want to update the bios if there's no new bios available? Keep in mind that the manufacturer might patch vulns in the bios and not tell you about it.

peteruithoven, 2017-11-25:
The Register article on this: https://www.theregister.co.uk/2017/11/20/intel_flags_firmware_flaws/

My laptop seems vulnerable ( I ran the detection program ) and there is no firmware/bios upgrade available yet. 
I'm still reading up, but would me_cleaner (partially) fix the vulnerability? 
It might be good to add something to the readme and or wiki?

Reading this topic it seems smart to first run me_cleaner before upgrading firmware/bios?

platomav, 2017-11-25:
> My laptop seems vulnerable ( I ran the detection program ) and there is no firmware/bios upgrade available yet.

Check the [Intel Management Engine Drivers, Firmware & System Tools](https://www.win-raid.com/t596f39-Intel-Management-Engine-Drivers-Firmware-amp-System-Tools.html#msg10191) thread for updates. Pay attention to the warnings.

> I'm still reading up, but would me_cleaner (partially) fix the vulnerability?

Not really

> Reading this topic it seems smart to first run me_cleaner before upgrading firmware/bios?

The opposite

corna, 2017-11-25:
As @platomav pointed out you can manually download a patched version of Intel ME (from his thread) and apply the fix by yourself.

sakaki-, 2017-11-27:
@platomav, apologies, I'm probably being a bit dense here; but could you please elaborate on your comment:

>> I'm still reading up, but would me_cleaner (partially) fix the vulnerability?

> Not really

If `me_cleaner` (in `-S` or no-option usage) purges the majority of the ME's firmware modules so it cannot successfully start up, how would an ME-related exploit still function? I appreciate there may be orthogonal BIOS firmware vulnerabilities as well, but at least some of the Intel CVEs cited in [the Register article](https://www.theregister.co.uk/2017/11/20/intel_flags_firmware_flaws/) @peteruithoven mentioned appear to relate to the ME only (kernel etc.)

Or, are you saying that since the kernel is not one of firmware modules purged by `me_cleaner` (for ME >= 11), it (i.e., the ME's kernel) is still running even on a 'cleaned' system, and thereby can still be attacked via buffer overflow etc.?

platomav, 2017-11-27:
> Or, are you saying that since the kernel is not one of firmware modules purged by me_cleaner [...]

Exactly. Specifically bup, kernel, rbe & syslib which are the most important and non-removable modules. It could be that running me_cleaner bricks the CSE enough to probably not be able to be operational and thus attacked (depending on how the vulnerabilities actually work, no details yet) but since the main modules are left intact, they can theoretically be targeted. So, me_cleaner is not a substitute of a proper INTEL-SA-00086 fix. The proper thing to do, if you want to use me_cleaner, is to use it on non-vulnerable firmware (CSME 11.8, 11.11 & 11.21, CSTXE 3.1 & 3.3, CSSPS 04.xx.04.xxx) and then flash that back.

persmule, 2018-07-20:
> And Intel won't release patches for Sandy Bridge/Ivy Bridge processors. As I understand me_cleaner can't guarantee that it prevents such CVEs? 

me_cleaner removes most ME modules, including those provide the "HTTP handler", thus already preventing this CVE, so you should keep using rather than discarding your precious T520.

pedrib, 2018-07-20:
> My Thinkpad T520 uses a Core i5 (Sandy Bridge), so can I throw it away now? I'm fed up with the "we don't support your hardware anymore, it is your fault now" policy. We really need libre processor architecture in silicon to overcome these issues.

Definitely need that open hardware / firmware. Looks like computers are going the same way as mobile phones, dump it every 3 years and get a new one. What a waste...

persmule, 2018-07-20:
@niclashoyer for your T520, you could even use [coreboot](https://www.coreboot.org/Board:lenovo/t520) on it.

niclashoyer, 2018-07-20:
I'll try coreboot using me_cleaner. Writing this from a Gigabyte B75M-D3H, which is supported by coreboot, too.