corna/me_cleaner #102: Intel Core i5 3570K Ivy Bridge on Biostar Motherboard

Sbalian02, 2017-11-21:
CPU Architecture = Ivy Bridge
CPU Model = Intel Core i5 3570K (Not Overclocked)
Custom Desktop with Biostar H61MLV Version 7.0 Motherboard - Intel H61 Chipset
OEM BIOS
I tried all three options in me_cleaner: No Flag, -S, and -s

Unfortunately, me_cleaner does not appear to work on my system. I was able to connect to my BIOS chip (Winbond W25Q32.V in DIP-8 package) using a Raspberry Pi, and used flashrom to dump an image. I ran me_cleaner on the dump, got the "Done! Good luck!" message, and was able to successfully flash the modified firmware onto the BIOS chip. However, when I plugged the desktop back in and tried to boot, it went into a reboot cycle about every five seconds. I made copies of the original dump and ran me_cleaner, once with -s and once with -S, and took turns flashing those modified images. The result was always the same, the system would just fall into the cycle of rebooting about every five seconds. I made sure there were no read/write errors before each attempt at flashing the BIOS chip by dumping two copies of whatever was on the chip and comparing hashes, so I have to conclude that for whatever reason, me_cleaner does not work on this crappy Biostar setup.

As a silver lining, at the end I was able to flash the very first dump I had made, and my desktop booted normally - which is better than a bricked system. However, the Management Engine is still alive and kicking. :

[user@system intelmetool]$ sudo ./intelmetool -s
[sudo] password for user:
Bad news, you have a H61 Express Chipset Family LPC Controller so you have ME hardware on board and you can't control or disable it, continuing...

MEI not hidden on PCI, checking if visible
MEI found: [8086:1c3a] 6 Series/C200 Series Chipset Family MEI Controller #1

ME Status : 0x1e000245
ME Status 2 : 0x60000006

ME: FW Partition Table : OK
ME: Bringup Loader Failure : NO
ME: Firmware Init Complete : YES
ME: Manufacturing Mode : NO
ME: Boot Options Present : NO
ME: Update In Progress : NO
ME: Current Working State : Normal
ME: Current Operation State : M0 with UMA
ME: Current Operation Mode : Normal
ME: Error Code : No Error
ME: Progress Phase : Host Communication
ME: Power Management Event : Clean Moff->Mx wake
ME: Progress Phase State : Host communication established

ME: Extend SHA-256: 0ed38a133a458b391f17f4277a6429f26073b6565280ad74b565bb936bc76a48

ME: timeout waiting for data: expected 8, available 0
ME: GET FW VERSION message failed
ME Capability: Full Network manageability : OFF
ME Capability: Regular Network manageability : OFF
ME Capability: Manageability : OFF
ME Capability: Small business technology : OFF
ME Capability: Level III manageability : OFF
ME Capability: IntelR Anti-Theft (AT) : OFF
ME Capability: IntelR Capability Licensing Service (CLS) : ON
ME Capability: IntelR Power Sharing Technology (MPC) : ON
ME Capability: ICC Over Clocking : ON
ME Capability: Protected Audio Video Path (PAVP) : ON
ME Capability: IPV6 : OFF
ME Capability: KVM Remote Control (KVM) : OFF
ME Capability: Outbreak Containment Heuristic (OCH) : OFF
ME Capability: Virtual LAN (VLAN) : OFF
ME Capability: TLS : OFF
ME Capability: Wireless LAN (WLAN) : OFF

Also, here is the output when I run me_cleaner on the original dump I made:

[user@system me_cleaner]$ python me_cleaner.py -S -O modified_firmware.bin original_firmware.bin
Full image detected
The ME/TXE region goes from 0x1000 to 0x200000
Found FPT header at 0x1010
Found 11 partition(s)
Found FTPR header: FTPR partition spans from 0x38000 to 0xa2000
ME/TXE firmware version 7.0.10.1203
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x05)...
Reading FTPR modules list...
 UPDATE           (LZMA   , 0x078fad - 0x07903f): removed
 BUP              (Huffman, fragmented data    ): NOT removed, essential
 KERNEL           (Huffman, fragmented data    ): removed
 POLICY           (Huffman, fragmented data    ): removed
 HOSTCOMM         (LZMA   , 0x07903f - 0x07e5d5): removed
 RSA              (LZMA   , 0x07e5d5 - 0x081128): removed
 CLS              (LZMA   , 0x081128 - 0x085ade): removed
 TDT              (LZMA   , 0x085ade - 0x08bba6): removed
 FTCS             (Huffman, fragmented data    ): removed
The ME minimum size should be 303104 bytes (0x4a000 bytes)
The ME region can be reduced up to:
 00001000:0004afff me
Setting the AltMeDisable bit in PCHSTRP10 to disable Intel ME...
Checking the FTPR RSA signature... VALID
Done! Good luck!

corna, 2017-11-25:
Unfortunately there's not much we can do: it seems that your BIOS waits for Intel ME and, if it doesn't reply, it forcefully reboots. Sadly, this is sadly, as BIOS vendors often make the assumption that Intel ME will be always present, and doesn't implement correct checks of its presence during the initialization of the system.

You could try to ask them to implement proper checks and timeouts, but honestly I'm afraid you won't be able to obtain much.

closed 2017-11-25 by corna

Sbalian02, 2017-11-25:
Thanks for replying, Corna. I'll shoot off an email to Biostar, maybe a 1% chance they'll respond.

corna, 2017-11-25:
Ok, let me know if you receive a reply

Sbalian02, 2017-12-03:
Update: I used the E-support section on their website to post the following. Let's see if they respond. I'm not holding my breath, at best they might say "ME is required on your system for security, functionality, etc."

I have a desktop computer running on a Biostar H61MLV Version 7.0 Motherboard. I have the latest version of the BIOS installed, version H61SR305.BSS dated 2013-03-05. The BIOS includes version 7.0.10.1203 of the Intel Management Engine. The Management Engine is known to have been compromised and vulnerable to malicious attack, however there is a recently disclosed method which Intel includes in order to disable the Management Engine. I set the disable option using the me_cleaner software available on GitHub. However, it appears Biostar designed the BIOS such that, if the Management Engline is disabled, the system enters into a reboot cycle roughly every five seconds after being powered on.

I am writing to ask what the proper method is to disable the Managment Engine in the motherboard and BIOS version I listed above so that the system can be properly booted. Thank you.