corna/me_cleaner #103: Intel Boot Guard patch issue (Could not apply d5839e0)

peteruithoven, 2017-11-25:
I was trying to figure out whether my laptop has Intel Boot Guard enabled using the following documentation:
https://github.com/corna/me_cleaner/wiki/Intel-Boot-Guard

(I copied the latest cherry pick string)
```
$ git clone --depth=1 http://review.coreboot.org/p/coreboot
$ cd coreboot/
$ git fetch https://review.coreboot.org/coreboot refs/changes/28/16328/26 && git cherry-pick FETCH_HEAD
remote: Counting objects: 339899, done
remote: Finding sources: 100% (339899/339899)
remote: Total 339899 (delta 269711), reused 339882 (delta 269711)
Receiving objects: 100% (339899/339899), 75.72 MiB | 1.74 MiB/s, done.
Resolving deltas: 100% (269711/269711), done.
From https://review.coreboot.org/coreboot
 * branch            refs/changes/28/16328/26 -> FETCH_HEAD
error: could not apply d5839e0... util/intelmetool: Add bootguard information dump support
hint: after resolving the conflicts, mark the corrected paths
hint: with 'git add <paths>' or 'git rm <paths>'
hint: and commit the result with 'git commit'
```

The review.coreboot.org page also mentions "Cannot Merge".

corna, 2017-11-25:
Yes, it seems that the latest changes created a merge conflict
Just checkout instead of cherry-picking it: `git fetch https://review.coreboot.org/coreboot refs/changes/28/16328/26 && git checkout FETCH_HEAD`

closed 2017-11-25 by peteruithoven

corna, 2017-11-25:
They've just merged the patch in coreboot, I've updated the Intel BG wiki page.