corna/me_cleaner #104: "Unknown image" for Xiaomi Notebook Air 12.5

sknsean, 2017-11-25:
Hi, ( Thanks for the great work)

**ifdtool -d TR100A05.fd**
seems fine:
https://gist.github.com/anonymous/050cc2652ca61979c04aa56b24ce0f14

**./me_cleaner.py -c TR100A05.fd**
returns "Unknown image"

**My notebook info**
https://gist.github.com/anonymous/0d5f9880a4e69d81f3444b2c6f695466

I know i'm trying to "clean" a newer bios, than i'm currently have installed. I'm planning on first dumping my current bios, then "clean" it and try with the new one :-)

corna, 2017-11-28:
`ifdtool -d` is not fine, see the lines 102-112: are your BIOS and ME firmware just 4kB? ;)
Somehow `ifdtool` considered the input image valid, while it is obviously not. It seems that this happens sometimes, I'm adding a section in the wiki to check whether the output of `ifdtool` makes sense.

You need a valid image, better if dumped from your PC (to keep things like serial and MAC address).

skny-prevas, 2017-11-28:
Thanks.
I will dump and try again (and of course report back) :-)

sknsean, 2017-11-28:
Thanks.
I will dump and try again (and of course report back) :-)

corna, 2017-11-28:
Can I copy your gist and use it as "bad ifdtool output" example?

sknsean, 2017-11-28:
Yes 👍

corna, 2017-11-28:
Thanks.
Wiki updated