corna/me_cleaner #105: "Unknown Image" for Asus P6T SE

zyliwax, 2017-11-29:
This is really not all that surprising given the outputs I've received, but this platform does seem to have Intel ME as would any Nehalem platform:

```
$ ./intelmetool -dm
Bad news, you have a `82801JIR (ICH10R) LPC Interface Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI not hidden on PCI, checking if visible
MEI device not found
```

Similar results came from `ifdtool`:

```
$ ./ifdtool -d p6t_se.rom
File p6t_se.rom is 2097152 bytes
No Flash Descriptor found in this image
```

Finally, this result from `me_cleaner` itself should be unsurprisng:

```
$ ./me_cleaner.py p6t_se.rom
Unknown image
```

It is worth noting this output from `file` since it seems to indicate some relevant information as to why these procedures may be failing: 

```
$ file p6t_se.rom
p6t_se.rom: DOS executable (block device driver)
```

Based on the output of these commands, I am led to conclude that my platform does indeed contain Intel ME, but that the available tools cannot remove (much less access) its contents.

I am hopeful this is simply a case of the BIOS image simply needing conversion into another format. I extracted the BIOS originally by physically removing the chip and reading its contents using a CH341A programmer, done several times to ensure that the image was properly extracted. Interestingly, these results are all the same when run against the latest BIOS image available from ASUS's website (https://www.asus.com/Motherboards/P6T_SE/HelpDesk_BIOS/). With any luck, I would like to completely remove the ME firmware as the information from (https://www.win-raid.com/t2443f39-Remove-ME-Ignition-firmware-completely-amp-Panther-Point-PCH-observations.html#msg33301) suggests is possible on Nehalem platforms.

Any helpful input or suggestions on where to proceed from here would be warmly appreciated, and of course I am happy to contribute my findings back to the community.

skochinsky, 2017-11-29:
The chipset possibly contains ME*, but the linked download contains only BIOS (legacy 16-bit AMIBIOS, pre-UEFI). You can try dumping the flash chip to double-check but it seems there is no ME firmware present in the flash, so there is nothing to clean.

* it seems ICH10R is positioned as a "consumer" variant and is not intended for AMT usage, so quite possibly ME is even removed or fused off completely, but there is a small possibility that the ME boot ROM still runs to perform the very basic initialization but stops because of missing ME firmware.

platomav, 2017-11-29:
X58 ICH does not contain a ME co-processor and thus not ME firmware

zyliwax, 2017-11-30:
@skochinsky @platomav Thank you for your informative responses. I did do a cursory comparison of the hexdumps from both my ROM file extracted from flash and the ROM file downloaded from ASUS, which have some differences but are mostly the same. Searching for any actual ME firmware would require further independent research, and I imagine that would start with determining what offset(s) to look at and what sort of byte patterns to look for.

For the time being, I will take your assessment that this platform lacks ME firmware as valid and close this issue, but I would like to cross-reference it with regards to intelmetool since that would mean that its output is misleading.

Should there be some outstanding reason to re-open this issue, whomever has the power should feel free to do so.

closed 2017-11-30 by zyliwax

corna, 2017-11-30:
The output of `ifdtool` is the key here:

     $ ./ifdtool -d p6t_se.rom
     File p6t_se.rom is 2097152 bytes
     No Flash Descriptor found in this image

On ICH10 (and before) it is possible to disable completely Intel ME by using a descriptor-less image, an image where the is BIOS mapped to the whole flash, no regions at all.

Yes, the output of intelmetool is misleading.