corna/me_cleaner #106: Question: running OEM BIOS update after external me_cleaner flash?

justaquickvisit, 2017-11-29:
I successfully ran me_cleaner on my original BIOS and flashed it back.

Now I need a BIOS upgrade to support new hardware.

What I want to know is if it is okay to use the BIOS flash utility and it will only overwrite the BIOS part of the BIOS or if it will overwrite the cleaned portion also.

skochinsky, 2017-11-29:
It seems the update contains only the Ec and Bios regions, so it should be safe to update. AFAIK Lenovo usually distributes ME updates separately.

justaquickvisit, 2017-11-29:
Very helpful!  How did you find that?  What tool did you examine the file with?

justaquickvisit, 2017-11-29:
I just flashed and intelmetool gives me good output according to this page:
https://github.com/corna/me_cleaner/wiki/External-flashing

corna, 2017-11-29:
As @skochinsky pointed out, once you've established that a BIOS update doesn't contain a new ME image, it shouldn't touch the ME region. In the worst case it could refuse to update himself due to the "damaged" ME.

closed 2017-11-29 by corna

skochinsky, 2017-11-29:
@justaquickvisit: I extracted the boot floppy image with IsoBuster and inspected the update files with a hex editor. They have a [$PFH footer](https://gist.github.com/skochinsky/181e6e338d90bb7f2693098dc43c6d54) with the list of regions in the update file, and I only saw "Bios" and "Ec".