corna/me_cleaner #107: [Success] ASRock X99E-ITX/ac, but...

cocafe, 2017-11-29:
Thanks for your hard work on removing Intel ME.

And I just used a CH341A programmer to dump and modify the BIOS SPI chip with '-S' option by me_cleaner.

And it works, on ASRock X99E-ITX/ac mainboard and Xeon E5-2680v4 CPU.

No 30 mins auto-shutdown issue, and ME PCI device disappeared.

**But, some features of UEFI do not work properly.**

I cannot adjust BCLK anymore, it does not work. 
And the system frequencies (BCLK related) are not as precise as before. (2499MHz -> around 2495MHz, causes by the clock PLL configuration, maybe)

And I cannot tweak some CPU features, something like: Turbo Boost 3.0 Tech is disabled in UEFI but it's indeed enabled in system.

I think this issue may cause by disabled ICC function in Intel ME.

Dunno soft-only disable bit will work better or not, will test and report later.

cocafe, 2017-11-29:
Both Instance Flash (internal flash) and Soft-Disable Bit Only method will cause this clocks side-effects.

Grimeton, 2017-11-29:
@cocafe So I checked. The problem here is that I'm running a Xeon E5-1620v4. It's not exactly overclocker friendly. I can change the BCLK to 101 MHz and I can then see the CPU going up and above 100MHz but going down again after a second. I guess that's Intel's lock on the speed?

Also the Turbo Boost 3.0 is not available with this CPU. But the feature is still there in the device manager and shown as problematic (yellow exclamation mark) as before. So no changes there.  

[image]

The memory adheres to the settings and is set to 1200 MHz (2400), by its XMP profile. So that's fine too.

Anything else I could look for or test to help you?

Grimeton, 2017-11-29:
@cocafe 
[image]

Note the PLL options at the bottom. Maybe something is wrong there?

corna, 2017-11-29:
You should try with `-s` and, if it works, with `-S -w MFS` to keep the MFS partition.

Grimeton, 2017-11-29:
@corna As a side note to the unknown image problem: ifdtool detects the image inside the updater's .exe file without problem but me_cleaner is not able to use it. So it's necessary to extract it correctly sometimes.

corna, 2017-11-29:
@Grimeton Are you sure the output of `ifdtool` makes sense? Sometimes it doesn't fail, but the output is only garbage; I've recently updated the [External flashing wiki page](https://github.com/corna/me_cleaner/wiki/External-flashing#dump-of-the-original-firmware) with some info about the `ifdtool` output, can you check it?

Grimeton, 2017-11-29:
@corna Here two screenshots with the two sections you mentioned on the wiki page.

Those are taken from the .exe file of the ASRock Bios Flash Tool that contains the ROM.

[image]

[image]

And they look pretty much the same as when I do the dump on the ROM directly.

cocafe, 2017-11-29:
@Grimeton 

Thanks for checking out that clock on your board.

I had ever increased BCLK a bit in UEFI, and it didn't drop.
Now it indeed lost control over BCLK stuffs.

If you disabled TB3.0 in UEFI, this Broken Device won't appear in system.
And that is what I disabled in NO-ME UEFI, and it still appears.

Just one more favor, Did your clocks change after removing ME?

cocafe, 2017-11-29:
@corna 

> You should try with -s and, if it works, with -S -w MFS to keep the MFS partition.

Nope, not having that luck, `-s` still will break clocking features. Is it worth to try out `-S -w MFS` ?

Grimeton, 2017-11-29:
@cocafe Ok, I can verify that disabling the Turbo Boost 3.0 feature in the bios does NOT remove the device from the device manager. But I can't really tell if this is a Bios problem or is caused by the fact that my CPU does not support it anyway as I do not have something to compare it to.

What I can tell is, that it was set to "Auto" and was always there. AFAIK the "Auto" setting should disable it as soon as an unsupported CPU is detected, which was never the case.

The clocks always change after reflashing as my board automagically loads setup defaults in the bios which changes the memory back to 2133 and so on. Is that what you mean by "the clocks have changed" ? 

Also don't forget that, at least my CPU, is constantly speed stepping:

[image]

Which results in rather odd values.

cocafe, 2017-11-29:
@Grimeton 

Dude, Mine does not have TB3.0, too. We both have asrock board, the same chip.
If you leave TB3.0 AUTO or ENABLED in UEFI, this device will appear. DISABLE it manually, it will dismiss.

And yeah, your CPU is dynamic clocking (the multiplier), but the clock can be different, it can be affected by the reference clock (BCLK). 

What I want to check out, is that whether your BCLK or overall system clocks (like DRAM) changed after removing ME. 

By the way, you can just decrease BCLK a bit, and then take a look at the CPU clock at **12x**, to verify your BCLK clocking still working or not.

I am gonna try again will new options, brb.

Grimeton, 2017-11-29:
So,

forgive me for asking, but the last time I overclocked something I changed the multiplier from 1x to 2x by setting a jumper and turning a 486DX33 into an DX2-66.

My board is not just changing the multiplier. It's also changing the bus speed. I "underclocked" it to 96MHz and it goes down (96x12) there but also goes up to 98.94MHz x 36 which gives me 3560.35MHz on the CPU. 

Also the single thread turbo is working when I stress the CPU so it goes up to 3.8GHz.

That all seems fine to me.

corna, 2017-11-29:
@Grimeton That ifdtool output is not valid:
 * 13 MiB of descriptor starting at 0xba000 (should be 4 KiB, starting at 0x0)
 * Meaningless region access 
    * for example the CPU cannot write its region, but it can write (but not read) the ME and descriptor regions

skochinsky, 2017-11-29:
One of the ME modules that usually gets removed by me_cleaner is ICC (internal clock controller) so it's no surprise clocking-related features may stop working if you mess with it.

Grimeton, 2017-11-29:
@corna yeah it takes the binary "garbage" in account that contains the code of the main executable.

cocafe, 2017-11-29:
@Grimeton @corna @skochinsky 

Thanks a lot, guys.

Tried flashing stuffs all night, no luck to make ICC work with ME stripped.

Maybe add to disadvantage in wiki descriptions to let people know this issue may occur on some platform? (Excuse me if it has already explained somewhere)

Anyway, appreciate for all the replies.

closed 2017-11-29 by cocafe

corna, 2017-11-29:
> Is it worth to try out `-S -w MFS`?

Nope