corna/me_cleaner #109: Great book on Intel Management Engine / AMT / vPRO (PDF/EPub)

IntelSucks, 2017-11-29:
Great work on me_cleaner.

Here is a detailed book on Intel ME, I'll leaving it here for other researchers and developers to find.

**Platform Embedded Security Technology Revealed**
**Safeguarding the Future of Computing with Intel Embedded Security and Management Engine**

PDF: https://link.springer.com/content/pdf/10.1007%2F978-1-4302-6572-6.pdf
EPUB: https://link.springer.com/download/epub/10.1007%2F978-1-4302-6572-6.epub

12 Chapters total.

Chapter 1:

This book is an in-depth technical introduction to an embedded system developed 
and manufactured by Intel Corporation. The embedded system is not an independent 
product; it is a native ingredient inside most of Intel’s computer product portfolio, which includes servers, desktops, workstations, laptops, tablets, and smartphones. 

Although not well known to most end users, the embedded system plays a critical role in many consumer applications that people use every day. As such, its architecture, implementation, and security features are worth studying.

Chapter 2:

When Intel’s Active Management Technology (AMT) first appeared in 2005, it was 
marketed as an advanced system management feature for Intel 82573E series gigabit 
Ethernet controllers. In 2007, a new embedded coprocessor, namely the management 
engine, was introduced. Originally, the management engine was designed primarily 
for implementing the AMT rather than running security applications. At that time, the main problem that was supposed to be resolved by the embedded engine and AMT was the high expense and difficulty of system management by network administrators. The management engine was a component of Intel chipsets with vPro technology. The Intel AMT implementation was moved from gigabit Ethernet controllers to the management engine and became a feature of vPro.

This book is not the first literature on the engine. Back in 2009, Intel Press published Active Platform Management Demystified: Unleashing the Power of Intel vPro Technology, authored by Intel’s Arvind Kumar, Purushottam Goel, and Ylian Saint-Hilaire.

It will be referred to as the “2009 AMT book” in this chapter.

skochinsky, 2017-12-30:
The 2009 AMT book is available here:

http://www.meshcommander.com/active-management

corna, 2018-01-05:
Thanks for the links.

closed 2018-01-05 by corna