corna/me_cleaner #110: Remove Intel ME/AMT from Intel network cards?

IntelSucks, 2017-11-29:
In Chapter 2 of Technology Revealed Platform Embedded Security

PDF: https://link.springer.com/content/pdf/10.1007%2F978-1-4302-6572-6.pdf
https://link.springer.com/book/10.1007/978-1-4302-6572-6

It is noted that AMT already existed back in 2005 in the 82573E Ethernet controller.

>When Intel’s Active Management Technology (AMT) first appeared in 2005, it was 
marketed as an advanced system management feature for Intel 82573E series gigabit 
Ethernet controllers.

A lot of people is wondering if their Intel network card is bugged to.

Is it possible for me_cleaner to support cleaning Intel network cards roms in the future?

skochinsky, 2017-11-29:
Where is this "lot of people"?

Do you have such a card and a flash dump from it?

Intel-Sucks, 2017-11-30:
>Where is this "lot of people"?

Who would pretend everyone still feel safe with Intel NICs hearing about vulnerabilities in their CPUs? Intel fan boys?

The Libreboot/Coreboot community for one don't trust Intel NICs:

>https://mail.coreboot.org/pipermail/coreboot/2017-November/085598.html

>I am looking for dual port gigabit and 10 gigabit PCI-e NIC's that 
aren't made by intel but don't require silly binary drivers or what not 
(ie: the closest thing to free) What would be my best option?

Here is one Intel NIC firmware:

>http://www.dell.com/support/home/us/en/19/Drivers/DriversDetails?driverId=6FD9P

>Intel NIC Firmware Family Version 16.5.0 for X520, X540, and I350 Adapters

If you can remote boot from an Intel NIC, you can also be remotely hacked from an Intel NIC (Intel Boot Agent/PROBOOT/ISBoot), who knows what backdoors are there just like their CPUs:

https://downloadcenter.intel.com/download/19186/Intel-Ethernet-Connections-Boot-Utility-Preboot-images-and-EFI-Drivers

>Intel® Ethernet Connections Boot Utility, Preboot Images, and EFI Drivers 

>The Intel® Ethernet Flash Firmware Utility (BootUtil) is a utility that you can use to program the PCI option ROM on the flash memory of supported PCI and PCIe*-based Intel® Network Adapters and to update configurations.

>BootUtil replaces older utilities and provides the functionality of the older **Intel Boot Agent, IBAUTIL; iSCSI Remote Boot**, ISCSIUTL; LANUTIL; and FLAUTIL utilities that were found in the PROBOOT, ISBoot, and, PROEFI web packs.

ilikenwf, 2017-12-08:
This kind of thing is why a lot of Libreboot folks just use Atheros cards.

my 2c:

That said even blob based cards, be it a MIPS broadcom netextreme 1/2 or an intel LAN or WLAN card should be relatively neutered if you strip the CPU me, and remove computrace and the IP stack from your bios (bnx2 is a special beast since it is exploitable, but only briefly during boot if I recall).

If your adversary has unlimited resources (government or giant company) you can maybe slow them down, and possibly stop them if you aren't worth the effort, otherwise most other adversaries should be shut down by what we're doing now.