corna/me_cleaner #112: Undo HAP bit setting

p4block, 2017-12-01:
Long story short, I have a modified BIOS that just had the HAP bit set (-s), and the original dump was lost and is not available to download anywhere.

I'd like to set the bit back to 0 to restore it to its original state for testing. I'm not confident enough in my python skills to hack the current py to do so hah

Great work on this project, my thanks to everyone involved.

corna, 2017-12-01:
Change [line 741](https://github.com/corna/me_cleaner/blob/2ff65c1ea72d2ba1ed9abe4d9bb0a079d79710b1/me_cleaner.py#L741):

     pchstrp0 |= (1 << 16)

to

     pchstrp0 &= ~(1 << 16)

and run `me_cleaner -s <image>`.

Check the result with `vbindiff`, only the byte at offset 0x102 (for ME > 11) should have changed.

For ME < 11 do the same on [line 748](https://github.com/corna/me_cleaner/blob/2ff65c1ea72d2ba1ed9abe4d9bb0a079d79710b1/me_cleaner.py#L748).

Wiggum127, 2017-12-01:
Instead of changing the source code, couldn't this become an option to set via the command line? So either set the HAP on or off?

p4block, 2017-12-01:
A thousand thanks @corna, worked like a charm.

Asus TP301UA has some weird setup, doesn't appear to have vboot enabled but disabling the ME even with -s completely kills the laptop.

Gonna do some probing with the latest intelmetool and confirm that there's no boot guard.

corna, 2017-12-01:
@Wiggum127 it's shouldn't be needed, people must always have a backup to rollback

@p4block unfortunately some BIOS are badly designed and doesn't handle well a disabled ME

closed 2017-12-01 by corna