corna/me_cleaner #116: [feature request] HAP bit test

tpruzina, 2017-12-04:
Can we add the functionality that tests HAP bit functionality by inserting "halt and catch fire" code into ME?
For example by inserting CLI (stop all interrupts) +HLT (halt CPU) into modules that we have control over.
This would serve as a kind of a validation that HAP bit actually works via simple litmus test - possibly bricking systems where this is not the case (hence why it should be accompanied with some big fat warning for potential users).

skochinsky, 2017-12-04:
>modules that we have control over.

These do not exist. The ME does not run *any* code which fails the signature check, whether the HAP bit is set or not, so patching any module achieves the same effect as removing it. Such code just won't be executed.

tpruzina, 2017-12-05:
Fair enough.

closed 2017-12-05 by tpruzina