corna/me_cleaner #118: Does not work on Z77X-UD3H

m4un6qilbi, 2017-12-05:
Hi everyone,

I tried to clean my Gigabyte Z77X-UD3H with me_cleaner.

Running <code>./me_cleaner.py -c original3.rom</code> gives

Full image detected
The ME/TXE region goes from 0x1000 to 0x200000
Found FPT header at 0x1010
Found 15 partition(s)
Found FTPR header: FTPR partition spans from 0x94000 to 0x109000
ME/TXE firmware version 8.1.30.1351
Checking the FTPR RSA signature... INVALID!!
The FTPR partition signature is not valid. Is the input ME/TXE image valid?

Running <code>./me_cleaner.py -S -d original3.rom -O modified.rom</code> gives

Full image detected
The ME/TXE region goes from 0x1000 to 0x200000
Found FPT header at 0x1010
Found 15 partition(s)
Found FTPR header: FTPR partition spans from 0x94000 to 0x109000
ME/TXE firmware version 8.1.30.1351
Reading partitions list...
 ???? (0x000003c0 - 0x000000400, 0x00000040 total bytes): removed
 FOVD (0x00000400 - 0x000001000, 0x00000c00 total bytes): removed
 MDES (0x00001000 - 0x000002000, 0x00001000 total bytes): removed
 FCRS (0x00002000 - 0x000003000, 0x00001000 total bytes): removed
 EFFS (0x00003000 - 0x00004b000, 0x00048000 total bytes): removed
 NVCL (NVRAM partition, no data, 0x00010511 total bytes): nothing to remove
 NVCP (NVRAM partition, no data, 0x0000a55b total bytes): nothing to remove
 NVJC (NVRAM partition, no data, 0x00005000 total bytes): nothing to remove
 NVKR (NVRAM partition, no data, 0x0001357f total bytes): nothing to remove
 NVSH (NVRAM partition, no data, 0x00007609 total bytes): nothing to remove
 NVTD (NVRAM partition, no data, 0x00001eac total bytes): nothing to remove
 GLUT (0x0004b000 - 0x00004d000, 0x00002000 total bytes): removed
 MDMV (0x0004d000 - 0x000093000, 0x00046000 total bytes): removed
 FTPR (0x00093000 - 0x000108000, 0x00075000 total bytes): NOT removed
 NFTP (0x00108000 - 0x00017d000, 0x00075000 total bytes): removed
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x8a)...
Reading FTPR modules list...
 UPDATE           (LZMA   , 0x0e36ff - 0x0e38be): removed
 ROMP             (Huffman, fragmented data    ): Traceback (most recent call last):
  File "./me_cleaner.py", line 719, in <module>
    args.keep_modules)
  File "./me_cleaner.py", line 332, in check_and_remove_modules
    offset, me_end)
  File "./me_cleaner.py", line 162, in remove_modules
    chunks_offsets = get_chunks_offsets(llut, me_start)
  File "./me_cleaner.py", line 115, in get_chunks_offsets
    i[1] = nonzero_offsets[nonzero_offsets.index(i[0]) + 1]
IndexError: list index out of range

What is the problem here?
Downloaded the file several times using an RPI3 and Flashrom.

Flash Chip is a Macronix MX25L6406E and i pulled \WP and \HOLD high using GPIOs.

Any Ideas what might be the problem here?

Thanks in advance

H2KChicken, 2017-12-05:
You may be able to use ifdtool to extract the ME partition and run me_cleaner over the extracted partition directly. Then you could insert the cleaned ME partition back into the rom and try testing with that.

subjector, 2017-12-05:
original3 ? Looks to me you did several dumps, did you verify any of them against the flash ?

flashrom --verify original3.rom

If the verify fails, there is a bad connection (check all wires or use shorter wires 10cm)

al3xtjames, 2017-12-06:
FWIW, Gigabyte doesn't bother implementing any forms of write protection, so you can internally flash with flashrom (make sure you have a backup!). This worked successfully on my GA-Z77X-UD5H (with `me_cleaner-S` on both the full image/extracted ME region).

m4un6qilbi, 2017-12-07:
Thanks for the quick answers.

The problem is I'm scared that there is something wrong with the stock ROM image and therefore that I would not have a working backup.
However I obtained the same Image 4 times.
Is there any way to check if there is something wrong with the Stock Image?

@H2KChicken
Is there any tutorial on doing this with ifdtool?
I'm supplying 3.3V through the PI, otherwise it would not recognize the Chip. Also the LEDs beneath the SPI-Flash do light up. However my PI keeps always crashing wenn i connect the power, so maybe the 3.3V Rail cannot provide enough current.

@al3xtjames
Unfortunately Gigabyte does not provide the ROM File for my Mainboard anymore, just the .exe Version. Did you extract that from the .exe or do you know any way how to do this
Is there

al3xtjames, 2017-12-07:
You should be able to extract the EXE with 7-Zip/binwalk/etc, I believe it is just a self-extracting archive.

subjector, 2017-12-07:
Do a time travel with the "wayback machine" they archive a lot. Visit the site , enter gigabyte url and jump to the desired date.

Also your pi might need a better powersupply (2A/2.5A) , could you elaborate on your pi setup. 

Did you run the --verify option after you dumped it ?

m4un6qilbi, 2017-12-09:
@al3xtjames 
Thanks. I could extract it from the exe. The Cleaner works on the vendor images.

@subjector 
I dumped 4 images and compared sha2 hashes. They are all the same. I guess --verify just verifies the image against the current read? So this should be the same or am I wrong?
I'm using the official powersupply (2,5A). Should I build up an external 3,3V supply using LDOs or is the stock one enough?

I'm a bit scared that the backups are broken.

corna, 2017-12-09:
@m4un6qilbi The 3.3V line of the RPi is provided by a (IIRC) 50 mA LDO, so the current capability of the power supply doesn't matter. Yes, you should use an external supply and connect its GND with the GND of the RPi.

However it doesn't explain why you keep getting the same broken image. Can you check if there's a repeating pattern (like all zeros after a point)? Try also to lower the spispeed parameter of flashrom.

Note that by using the ROM from the vendor's website you could miss some board specific bits (like the serial of the MB). Your board doesn't use the Intel GbE, so the MAC address isn't stored in the ROM file, but other things could be missing in the website's ROM.

subjector, 2017-12-09:
@m4un6qilbi
If the hashes are the same, the read is consistent. A bad connection/low voltage would give you random read errors. The flash or a flashregion could be protected. Is there actual data in your file ? (a file containing all FF or 00 will have a matching hash)