corna/me_cleaner #120: Intel Server Board with SPS 2.0 - me_clean says unknown image

atlury, 2017-12-06:
Friends,

I have an intel server board s2600jf whose firmware is here. Its a cap file.

hxxps://downloadcenter.intel.com/download/26952/Intel-Server-Board-S2600JF-Intel-Server-System-H2000JF-System-Update-Package-for-UEFI?product=56303

Below is Me Analyzer output

-------[ ME Analyzer v1.34.1 r106 ]-------
File:     MEComplete_02.01.07.328_HFW.cap (1/1)
Family:   SPS
Version:  02.01.07.328
Release:  Production
Type:     Region
Date:     2013-09-19
Size:     0x1C0000

I tried UEFItool to extract from cap file but both ifdtool and me_cleaner say unknown image.

[image]

skochinsky, 2017-12-06:
Intel capsules use a pretty weird deeply nested structure and you can't use them for flashing. Extract the dump from the flash chip and try that.

atlury, 2017-12-06:
Thanks @skochinsky shall solder to the chip and try it on the dump. Quick question, the board seems to have the first release of bios. Should i just update the me, bios, bmc and then try it? Or just try it?

corna, 2017-12-09:
Update it before, so that you won't have to re-apply me_cleaner after the update.

skochinsky, 2017-12-09:
The BIOS version should not matter, but if you update the ME it's better to update the BIOS too just in case.

atlury, 2017-12-09:
the problem is there are two winbond ICs. One named BMC and another named BIOS. wonder which one will contain the ME.

skochinsky, 2017-12-09:
most likely the BIOS one. BMC is a separate ARM CPU on this board.

atlury, 2017-12-10:
Thank you @skochinsky 

Will try in a few days and let you know.

gavinhungry, 2018-01-08:
@atlury,

Anything to report?

atlury, 2018-01-08:
@gavinhungry 
I havent tried soldering it yet. 

Even before that, I have been trying to run this board via a stand-online ATX psu. It doesnt work. That stupid BMC does not allow the board to post. It only allow the board to post with server power supplies with a back-plane etc.

I have to get rid of this BMC and get this board to accept ordinary ATX PSU.

[image]

atlury, 2018-01-08:
@gavinhungry 

That is the picture of the board where it does not post. Just amber light. BTW the ME engine works and there is an IP to it. Nevertheless must find a way to first neutralize BMC first.