corna/me_cleaner #122: Lenovo B51-80 works only with HAP

An4cl3t0, 2017-12-06:
Hi, I tried to apply me_cleaner on:
 
-Model: Lenovo B51-80
-CPU: i5-6200U
-BIOS/UEFI version: 11.x

Tried with both -s/-S, works only with soft disable (-s). Otherwise the PC won't boot. No screen. I'm happy with this, if you want more informations for debugging I'm glad to give you.

rsalvaterra, 2017-12-07:
Good stuff! I have the exact same machine. Is the SPI flash easily accessible?

An4cl3t0, 2017-12-07:
Not so much. The SPI is over the wireless card, you can see it removing the back panel, but you have to cut the plastic or disassembly to access with the clamp.

al3xtjames, 2017-12-07:
I do not know about that specific model, but most Skylake (and newer) Lenovo sytems have Boot Guard enabled (see (the wiki page)[https://github.com/corna/me_cleaner/wiki/Intel-Boot-Guard] for some more info), which could be why `-S` is failing.

An4cl3t0, 2017-12-08:
Thanks. This is probably the problem. Closed.

closed 2017-12-08 by An4cl3t0