corna/me_cleaner #123: AsRock Z97 Exterme 4 fails with -s/-S no BIOS w/o flags

eNTi, 2017-12-07:
- Haswell
- Intel Core i5-4670K CPU
- AsRock Z97 Exterme 4
- OEM BIOS
- No -s/-S flag in me_cleaner
```
MEI not hidden on PCI, checking if visible
MEI found: [8086:8cba] 9 Series Chipset Family ME Interface #1

ME Status   : 0x1e003052
ME Status 2 : 0x16322152

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : NO
ME: Manufacturing Mode      : YES
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Recovery
ME: Current Operation State : M0 with UMA
ME: Current Operation Mode  : Normal
ME: Error Code              : Image Failure
ME: Progress Phase          : BUP Phase
ME: Power Management Event  : Pseudo-global reset
ME: Progress Phase State    : M0 kernel load

ME: Extend SHA-256: d3d8c0553297e0037b559e55505156f3893755b01b49ba8c4e46e86157b35bbe

ME: has a broken implementation on your board with this BIOS
ME: failed to become ready
ME: failed to become ready
ME: GET FW VERSION message failed
ME: failed to become ready
ME: failed to become ready
ME: GET FWCAPS message failed
```
My biggest issue is that I can no longer access my bios menu to for example set turbo, cpu frequencies and the like. My monitor is just completely blanked until grub shows up. Is that by design?

So I tried -s and -S. Both flags give me:

```
MEI was hidden on PCI, now unlocked
MEI found: [8086:8cba] 9 Series Chipset Family ME Interface #1

ME Status   : 0x1e020191
ME Status 2 : 0x164d2142

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : NO
ME: Manufacturing Mode      : YES
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Initializing
ME: Current Operation State : Bring up
ME: Current Operation Mode  : Debug
ME: Error Code              : No Error
ME: Progress Phase          : BUP Phase
ME: Power Management Event  : Pseudo-global reset
ME: Progress Phase State    : 0x4d

ME: Extend SHA-256: d3d8c0553297e0037b559e55505156f3893755b01b49ba8c4e46e86157b35bbe

ME: failed to become ready
ME: failed to become ready
ME: GET FW VERSION message failed
ME: failed to become ready
ME: failed to become ready
ME: GET FWCAPS message failed
Re-hiding MEI device...done
```
So I guess I'm out of luck? Either forego any BIOS or IME? Should I try to install coreboot?

skochinsky, 2017-12-07:
overclocking features likely require the  ICC (_internal clock controller_) module from the ME, and if you remove it they won't work. Possibly other BIOS features depend on the fully functioning ME too.

eNTi, 2017-12-07:
Is this a tradeoff I have to live with then or are there other options?