corna/me_cleaner #124: ASRock Z97 Extreme6 - should it be working?

fabertawe, 2017-12-07:
I have a desktop with an ASRock Z97 Extreme6 mobo and an i7 4790k CPU. The "_me_cleaner status_" page shows someone with the identical MB and CPU having had success but I'm not.

I used uefitool to "extract the body" from the last BIOS available for this board (Z97Ex62.70) and called it 'bios.bin'. All the logs are below.

I had to add the kernel parameter 'iomem=relaxed' to be able to use intelmetool.

I tried _me_cleaner_ with and without the **-S** flag. I also tried using the Windows software (UBU) referenced in the successful flashing on the "_me_cleaner status_" page.

Lastly, I tried flashing a previous BIOS version (with reset to defaults) before attempting to flash my modified version again (twice and with reset to defaults).

Any help appreciated, thanks. 

**ifdtool -d bios.bin** 
```
File bios.bin is 8388608 bytes
FLMAP0:    0x03040003
  NR:      3
  FRBA:    0x40
  NC:      1
  FCBA:    0x30
FLMAP1:    0x15100206
  ISL:     0x15
  FPSBA:   0x100
  NM:      2
  FMBA:    0x60
FLMAP2:    0x00210120
  PSL:     0x2101
  FMSBA:   0x200
FLUMAP1:   0x00000cdf
  Intel ME VSCC Table Length (VTL):        12
  Intel ME VSCC Table Base Address (VTBA): 0x000df0

ME VSCC table:
  JID0:  0x0000471f
    SPI Componend Device ID 1:          0x00
    SPI Componend Device ID 0:          0x47
    SPI Componend Vendor ID:            0x1f
  VSCC0: 0x20152015
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x06
    Lower Write Status Required:        No
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    4KB
    Upper Erase Opcode:                 0x20
    Upper Write Enable on Write Status: 0x06
    Upper Write Status Required:        No
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    4KB
  JID1:  0x001740ef
    SPI Componend Device ID 1:          0x17
    SPI Componend Device ID 0:          0x40
    SPI Componend Vendor ID:            0xef
  VSCC1: 0x20252025
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x50
    Lower Write Status Required:        No
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    4KB
    Upper Erase Opcode:                 0x20
    Upper Write Enable on Write Status: 0x50
    Upper Write Status Required:        No
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    4KB
  JID2:  0x0016701c
    SPI Componend Device ID 1:          0x16
    SPI Componend Device ID 0:          0x70
    SPI Componend Vendor ID:            0x1c
  VSCC2: 0x20052005
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x50
[... 237 lines cut ...]
ME: Progress Phase          : Host Communication
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : Host communication established

ME: Extend SHA-256: 270338250cca1d439e1e38229776ad740360322ebaa0afc5e355438989032796

ME: Firmware Version 9.1.1005.25 (code) 9.1.1005.25 (recovery) 9.1.1010.2 (fitc)

ME Capability: Full Network manageability                 : OFF
ME Capability: Regular Network manageability              : OFF
ME Capability: Manageability                              : OFF
ME Capability: Small business technology                  : OFF
ME Capability: Level III manageability                    : OFF
ME Capability: IntelR Anti-Theft (AT)                     : OFF
ME Capability: IntelR Capability Licensing Service (CLS)  : ON
ME Capability: IntelR Power Sharing Technology (MPC)      : OFF
ME Capability: ICC Over Clocking                          : ON
ME Capability: Protected Audio Video Path (PAVP)          : ON
ME Capability: IPV6                                       : OFF
ME Capability: KVM Remote Control (KVM)                   : OFF
ME Capability: Outbreak Containment Heuristic (OCH)       : OFF
ME Capability: Virtual LAN (VLAN)                         : ON
ME Capability: TLS                                        : OFF
ME Capability: Wireless LAN (WLAN)                        : OFF
```

VonManGuy, 2017-12-07:
@GenericHero was able to succeed, maybe he could dump his BIOS flash.

subjector, 2017-12-07:
Afaik you need to flash with an external flasher (pi/ch341a), you cannot use the internal programmer. If you cross ref. the read/write access you'll understand why.

Could you elaborate on your setup for flashing ?

fabertawe, 2017-12-08:
@VonManGuy - there's no email address unfortunately.

@subjector - apologies for not adding the information originally. I am flashing in the BIOS via the updater there. @GenericHero used this option also and succeeded. I will look at the external option again but not sure if I want to get that hands on with the hardware!

corna, 2017-12-10:
An external programmer is unfortunately the only reliable way to flash a modified image: it may seems scary, but it is quite easy in reality.

`ifdtool -u` unlocks the RW access, but the resulting image must be flashed with an external programmer to apply the modification.

corna, 2017-12-10:
@platomav it seems that this board doesn't have a secondary chip, do you have any idea from where it's fetching the original ME image? It's a 2MB ME image, so a backup image could fit in the BIOS region.

Aalexeey, 2022-12-30:
Hi, any body has a bios bin dump of AsRock Z77 Extreme 6? I've get this mobo with old bios incompatible with broadwell cpu, so mobo doesn't starts. The flash is flashed successfully in willem programmer, but ASRock has modified bios parts relocation during AFUWIN or AFUDOS flash, so programmer dosn't know how to relocate. Need manual dumped bios bin or rom not from ASRock site. I'm trying to avoid a second way with buying a compatible CPU just for flashing.