corna/me_cleaner #126: AsRock Z77 Extreme 6 - ME region restores after flashing

jump111, 2017-12-10:
I used external flasher to write cleaned bios image on chip, but after booting with it I get "updating Intel ME" message from UEFI and binary comparison of subsequent dump reveals that ME region is restored - it's not what was in the image I flashed. Any suggestions how to prevent ME restoration on my motherboard?

corna, 2017-12-10:
Does your board have two separate BIOS chips?

skochinsky, 2017-12-10:
Sometime the BIOS may include a copy of the ME firmware and reflash it. It was particularly the case for some boards that could be upgraded from 7.1(?) to 8.0. You could try finding that image in the UEFI part (e.g. search for "$FPT") and replacing it with the cleaned one. @platomav may know more.

corna, 2017-12-10:
(from [#124](https://github.com/corna/me_cleaner/issues/124#issuecomment-350557964)) @platomav it seems that this board doesn't have a secondary chip, do you have any idea from where it's fetching the original ME image? It's a 2MB ME image, so a backup image could fit in the BIOS region.

platomav, 2017-12-10:
Yes, I know. You can use ME Analyzer with -enuf (enable UEFIFind) parameter to see. It is GUID B3160739-1365-48A7-AECB-038652E2B528.

[image]

Note that on some ASRock boards, there was an additional check at AMITSE module with hard-coded values for expected firmware version to be restored. If that fails, you'll end up in a boot loop. So make sure you can restore via a programmer. In that case, you can mod AMITSE with a hex editor (difficult) or use a dissasembler like IDA to find where the check occurs. I cannot do the latter but can the former, provided that the build number is unique enough.

jump111, 2017-12-11:
Well, the board ended up in boot loop. Interesting thing is that ME region is restored despite setting it to readonly in descriptor.
Is AMITSE GUID B1DA0ADF-4F77-4070-A88E-BFFE1C60529A?
platomav, my ME version matches version on your screenshot. Only size of cleaned image is different - 0x108000.

platomav, 2017-12-11:
The ME firmware at the Flash Descriptor (Engine region) and inside the GUID (BIOS region) are identical with the only difference being the GUID header at the latter. Provided that AMITSE checks the version only, you don't have to mess with it. Just make sure you replace the ME firmware properly twice. The size must be the same (0x1FF000) so add padding at me_cleaner's result if required. For the GUID, replace everything after its header (0x1C), the total size should be 0x1FF000 + 0x1C = 0x1FF01C.

skochinsky, 2017-12-11:
> ME region is restored despite setting it to readonly in descriptor.

The BIOS can send a special message asking ME to unlock its region so it can be reflashed (regardless of descriptor permissions). This works only until reset but does allow the BIOS to reflash the ME.

jump111, 2017-12-11:
Sizes and headers of original and clean B3160739-1365-48A7-AECB-038652E2B528 GUID match, but board still goes to boot loop. Guess UEFI is checking something else beside version.

platomav, 2017-12-12:
There might be other checks at AMITSE but that requires dissasembly so I don't have something else to suggest.