corna/me_cleaner #128: DG45ID unknown image isue

lasuelle, 2017-12-12:
Using UEFI tool ROM was extracted from bios image IDG4510H.86A.0135.EB.EXE which works on DG45ID motherboard.

After running me_cleaner.py command the error presented was "Unknown Image"

Looking for a workaround.

persmule, 2017-12-12:
I don now know whether me_cleaner support the combination G45/Ich10, and ME firmware could be removed completely. You could ask coreboot guys for that.
Besides, me_cleaner only works on raw ME image and raw boot rom image (i.e. what is directly read from boot rom flash). It cannot be guaranteed that what you extract from IDG4510H.86A.0135.EB.EXE is either of them.

lasuelle, 2017-12-12:
ok so what are the options here?

platomav, 2017-12-12:
This is the firmware that your board uses:

[image]

Intel .BIO images cannot be modified and then flashed back via official tools so your only option is to dump your SPI chip and remove the ME region entirely as that works for ME 2-5 and 6 Ignition. Provided that Intel's BIOS does not wait indefinitely for the ME to initialize, it should work. You need a programmer though.

lasuelle, 2017-12-13:
Programmer like this CH341A USB bios board programmer 24 25 series ?

lasuelle, 2017-12-13:
i hope this is all i need
1. SOIC 8 clip + adaptor
2. Bios Board MX25L6405

sakaki-, 2017-12-14:
I have a step-by-step guide with photos covering the full process [here](https://wiki.gentoo.org/wiki/Sakaki%27s_EFI_Install_Guide/Disabling_the_Intel_Management_Engine) (RPi3 based). You don't have to follow this exactly (for example, you can use Raspbian rather than Gentoo on your RPi if you want), but it will hopefully clarify the end-to-end flow.

lasuelle, 2017-12-15:
I'm using a non inbuilt NIC as a simpler fix https://github.com/corna/me_cleaner/issues/132

lasuelle, 2018-02-03:
Ok i found a proper fix. 
Go to
BIOS-->Advanced-->Intel ME setup prompt 
Hit Enable.
Reboot
After BIOS posts, Intel ME setup screen will appear 
Hit Ctrl+p to enter
go to "Intel ME state" and select disable
Reboot
Go to BIOS and Disable Intel ME Setup Prompt
Exit bios saving settings.
That is it.