corna/me_cleaner #129: FAIL: MSI X299 RAIDER - no POST, black screen, no boot

peter137, 2017-12-13:
CPU Architecture: **Skylake-X**
CPU model: **i7 7820X**
Laptop/motherboard: **MSI X299 RAIDER**
OEM BIOS or coreboot?: **OEM 7A94v16**
If you used the -s/-S flag in me_cleaner: no, command was: me_cleaner.py -O modified_firmware.bin original_firmware.bin  (no such option exists)

After updating BIOS Flash with modified frimware the $SUBJ happened, system was unusable. Backup BIOS brought back life, but would still love to remove the ME code entirely.

During error state MB Debug LEDs: Red, white (CPU, DRAM) light up.

**Pre-flight test:**
intelmetool -m
Could not get RCBA address
Error reading RCBA

**Validity check:**
me_cleaner.py -c ~/BIOS/7A94v16/E7A94IMS.160
Full image detected
The ME/TXE region goes from 0x3000 to 0x300000
Found FPT header at 0x3010
Found 11 partition(s)
Found FTPR header: FTPR partition spans from 0x4000 to 0xab000
Found FTPR manifest at 0x4448
ME/TXE firmware version 11.10.0.1287
Checking the FTPR RSA signature... VALID

**Cleaning log:**
me_cleaner.py -O ~/BIOS/7A94v16/E7A94IMS.160-mod ~/BIOS/7A94v16/E7A94IMS.160
Full image detected
The ME/TXE region goes from 0x3000 to 0x300000
Found FPT header at 0x3010
Found 11 partition(s)
Found FTPR header: FTPR partition spans from 0x4000 to 0xab000
Found FTPR manifest at 0x4448
ME/TXE firmware version 11.10.0.1287
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x84)...
Reading FTPR modules list...
 FTPR.man     (uncompressed, 0x004448 - 0x005018): NOT removed, partition manif.
 rbe.met      (uncompressed, 0x005018 - 0x0050ae): NOT removed, module metadata
 kernel.met   (uncompressed, 0x0050ae - 0x00513c): NOT removed, module metadata
 syslib.met   (uncompressed, 0x00513c - 0x0051a0): NOT removed, module metadata
 bup.met      (uncompressed, 0x0051a0 - 0x00574a): NOT removed, module metadata
 pm.met       (uncompressed, 0x00574a - 0x0057f8): NOT removed, module metadata
 vfs.met      (uncompressed, 0x0057f8 - 0x006158): NOT removed, module metadata
 evtdisp.met  (uncompressed, 0x006158 - 0x0062e6): NOT removed, module metadata
 loadmgr.met  (uncompressed, 0x0062e6 - 0x00640e): NOT removed, module metadata
 busdrv.met   (uncompressed, 0x00640e - 0x0067b4): NOT removed, module metadata
 gpio.met     (uncompressed, 0x0067b4 - 0x0068fe): NOT removed, module metadata
 prtc.met     (uncompressed, 0x0068fe - 0x006aae): NOT removed, module metadata
 policy.met   (uncompressed, 0x006aae - 0x006c72): NOT removed, module metadata
 crypto.met   (uncompressed, 0x006c72 - 0x006dfc): NOT removed, module metadata
 heci.met     (uncompressed, 0x006dfc - 0x006fc8): NOT removed, module metadata
 storage.met  (uncompressed, 0x006fc8 - 0x0072c4): NOT removed, module metadata
 pmdrv.met    (uncompressed, 0x0072c4 - 0x0073e8): NOT removed, module metadata
 maestro.met  (uncompressed, 0x0073e8 - 0x0074d2): NOT removed, module metadata
 fpf.met      (uncompressed, 0x0074d2 - 0x0075de): NOT removed, module metadata
 hci.met      (uncompressed, 0x0075de - 0x0076e0): NOT removed, module metadata
 fwupdate.met (uncompressed, 0x0076e0 - 0x0077ea): NOT removed, module metadata
 ptt.met      (uncompressed, 0x0077ea - 0x0078f6): NOT removed, module metadata
 touch_fw.met (uncompressed, 0x0078f6 - 0x007a40): NOT removed, module metadata
 rbe          (Huffman     , 0x007a40 - 0x00a100): NOT removed, essential
 kernel       (Huffman     , 0x00a100 - 0x019d40): NOT removed, essential
 syslib       (Huffman     , 0x019d40 - 0x02bc00): NOT removed, essential
 bup          (Huffman     , 0x02bc00 - 0x054880): NOT removed, essential
 pm           (LZMA/uncomp., 0x054880 - 0x056e80): removed
 vfs          (LZMA/uncomp., 0x056e80 - 0x05ec00): removed
 evtdisp      (LZMA/uncomp., 0x05ec00 - 0x060600): removed
 loadmgr      (LZMA/uncomp., 0x060600 - 0x063380): removed
 busdrv       (LZMA/uncomp., 0x063380 - 0x064c40): removed
 gpio         (LZMA/uncomp., 0x064c40 - 0x065ec0): removed
 prtc         (LZMA/uncomp., 0x065ec0 - 0x066a80): removed
 policy       (LZMA/uncomp., 0x066a80 - 0x06b580): removed
 crypto       (LZMA/uncomp., 0x06b580 - 0x079140): removed
 heci         (LZMA/uncomp., 0x079140 - 0x07d040): removed
 storage      (LZMA/uncomp., 0x07d040 - 0x081640): removed
 pmdrv        (LZMA/uncomp., 0x081640 - 0x0827c0): removed
 maestro      (LZMA/uncomp., 0x0827c0 - 0x084580): removed
 fpf          (LZMA/uncomp., 0x084580 - 0x085e00): removed
 hci          (LZMA/uncomp., 0x085e00 - 0x0866c0): removed
 fwupdate     (LZMA/uncomp., 0x0866c0 - 0x08b200): removed
 ptt          (LZMA/uncomp., 0x08b200 - 0x0a0380): removed
 touch_fw     (LZMA/uncomp., 0x0a0380 - 0x0ab000): removed
The ME minimum size should be 352256 bytes (0x56000 bytes)
The ME region can be reduced up to:
 00003000:00058fff me
Checking the FTPR RSA signature... VALID
Done! Good luck!

NoahtheLurker, 2017-12-15:
Did you consider trying any other method?

peter137, 2017-12-15:
Like SPI flashing? flasher toolkit is on its way, in a few weeks i'll try that too. Meanwhile any suggestions to try?

NoahtheLurker, 2017-12-15:
I would be pleased to see the results with the USB programmer and externally flashed BIOS.

peter137, 2017-12-16:
Ok, but lets make it clear, that the OEM flasher reported no errors.

corna, 2018-01-10:
You should try with the `-s` flag and, if it works, with `-S`.