corna/me_cleaner #131: Internal flashing dump command in linux?

lasuelle, 2017-12-13:
when attempting to take a bios chip dump under linux what commands are used when on OEM BIOS?

matt123b, 2017-12-18:
Please list your board model and BIOS version (version number and date if available). Not sure if I'll be able to help but I'm sure that others would find that info to be useful in helping you figure out a solution.

lasuelle, 2017-12-20:
DG45ID bios is version 135.

corna, 2018-01-10:
You can use _flashrom_:
`flashrom -p internal -r dump.bin`

Then _ifdtool_ to get the r/w permissions:
`ifdtool -d dump.bin`

The output will tell you if you can flash the ME region from the OS.

ArchangeGabriel, 2018-01-10:
You might even get an answer from the first command by getting an output line like this one:
```
FREG2: Warning: Management Engine region (0x00003000-0x006fffff) is locked.
```