corna/me_cleaner #132: Myth or fact about onboad NIC and ME?

lasuelle, 2017-12-13:
a while back there was a write up which suggested that if one can avoid the onboard NIC by installing an after market NIC it becomes possible to prevent ME from receiving any remote activation commands.

Whether this is a myth or is there some truth to it?

skochinsky, 2017-12-13:
Somewhat true; ME can only talk to Intel LAN chip (Ethernet or WiFi) and it must be *directly connected* to it, so you can either use an additional card or just don't connect anything to the onboard LAN socket (e.g. if you have two, don't use Intel one). That said, to get ME activated:

 - you need AMT firmware (usually corporate/business boards only)
 - you need an activation server on the LAN (maybe it even has to be on the same subnet, not sure)

It won't do anything on a random network not specifically set up for AMT.

lasuelle, 2017-12-15:
Thanks for the information.

closed 2017-12-15 by lasuelle