corna/me_cleaner #133: Dell Notebook (2011) in Managebility Mode " Disabled ME / 3 "

gizuone, 2017-12-15:
Hello,

first of all i want to apologise for my bad english. But i hope anyone can answer my question and maybe this informations are some kind of usefull.

After some Research i found out Dell is selling Notebooks and Desktops with "disabled" ME still some years ago. There exist Motherboards with Jumpers to "disable" the ME, but it was not mentioned in the handbooks and they send Notebooks in different Versions, including with ME disabled. Personaly for me, these infos were some kind of the holy grail and i happened to get an older Dell Notebook yesterday. It got this "Disabled ME 3" Sticker underneath the RAM Service Flap and after some tests i think it could replace my libreboot x200. What do you think ? It is still necessery to clean the IME ?

Attached some outputs.
```
root@debian:~/coreboot/util/intelmetool# ./intelmetool --me
Bad news, you have a `QS57 Chipset LPC Interface Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI found: [8086:3b64] 5 Series/3400 Series Chipset HECI Controller

ME Status   : 0xee110205
ME Status 2 : 0xb001f

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : YES
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Normal
ME: Current Operation State : Preboot
ME: Current Operation Mode  : (null)
ME: Error Code              : No Error
ME: Progress Phase          : ROM Phase
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : (null)

ME: Extend Register not valid

ME: timeout waiting for data: expected 8, available 6
ME: GET FW VERSION message failed
ME: timeout waiting for data: expected 5, available 0
ME: GET FWCAPS message failed

root@debian:~/coreboot/util/intelmetool# lspci
00:00.0 Host bridge: Intel Corporation Core Processor DRAM Controller (rev 02)
00:02.0 VGA compatible controller: Intel Corporation Core Processor Integrated Graphics Controller (rev 02)
00:19.0 Ethernet controller: Intel Corporation 82577LM Gigabit Network Connection (rev 05)
00:1a.0 USB controller: Intel Corporation 5 Series/3400 Series Chipset USB2 Enhanced Host Controller (rev 05)
00:1b.0 Audio device: Intel Corporation 5 Series/3400 Series Chipset High Definition Audio (rev 05)
00:1c.0 PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 1 (rev 05)
00:1c.1 PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 2 (rev 05)
00:1c.2 PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 3 (rev 05)
00:1c.3 PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 4 (rev 05)
00:1d.0 USB controller: Intel Corporation 5 Series/3400 Series Chipset USB2 Enhanced Host Controller (rev 05)
00:1e.0 PCI bridge: Intel Corporation 82801 Mobile PCI Bridge (rev a5)
00:1f.0 ISA bridge: Intel Corporation QS57 Chipset LPC Interface Controller (rev 05)
00:1f.2 RAID bus controller: Intel Corporation 82801 Mobile SATA Controller [RAID mode] (rev 05)
00:1f.3 SMBus: Intel Corporation 5 Series/3400 Series Chipset SMBus Controller (rev 05)
00:1f.6 Signal processing controller: Intel Corporation 5 Series/3400 Series Chipset Thermal Subsystem (rev 05)
02:00.0 Network controller: Broadcom Limited BCM43224 802.11a/b/g/n (rev 01)
03:00.0 SD Host controller: Ricoh Co Ltd MMC/SD Host Controller (rev 01)
3f:00.0 Host bridge: Intel Corporation Core Processor QuickPath Architecture Generic Non-core Registers (rev 02)
3f:00.1 Host bridge: Intel Corporation Core Processor QuickPath Architecture System Address Decoder (rev 02)
3f:02.0 Host bridge: Intel Corporation Core Processor QPI Link 0 (rev 02)
3f:02.1 Host bridge: Intel Corporation 1st Generation Core i3/5/7 Processor QPI Physical 0 (rev 02)
3f:02.2 Host bridge: Intel Corporation 1st Generation Core i3/5/7 Processor Reserved (rev 02)
3f:02.3 Host bridge: Intel Corporation 1st Generation Core i3/5/7 Processor Reserved (rev 02)
```
With dmesg i couldnt find any related.

Some Sources:
Preconfigured Notebooks and Versions:
https://community.spiceworks.com/topic/1396735-can-vpro-be-enabled-on-a-dell-laptop
https://hackaday.com/2016/11/28/neutralizing-intels-management-engine/#comment-3292119

Desktop Motherboard Jumper:
https://github.com/corna/me_cleaner/issues/3#issuecomment-287181243
https://www.heise.de/forum/heise-Security/News-Kommentare/Intel-Management-Engine-gehackt/ME-Firmware-On-Off-per-Jumper/posting-31085269/show/

mparnelldmp, 2017-12-15:
I personally would still clean the device and remove the IP stack and computrace modules from the BIOS myself...

gizuone, 2017-12-16:
I am currently diving into these Topics (computrace, IP stack) and how to remove these modules from BIOS. Fortunately the option to permanent disable Computrace was still choseable for me. For safety's sake i switched pxe and boot on lan off, too. Cleaning with ME_Cleaner is on the to-do list.
Thank you for the comment and the hints. :)

closed 2017-12-16 by gizuone