corna/me_cleaner #134: Sorta works on ThinkPad X230

matt123b, 2017-12-18:
So I've tested ME cleaner on several laptops that others have reported success with and I got the same results. I've also tried with an older MacBook and it worked well. So now I've moved on to my X230. I recently ran the BIOS update utility from Windows and upgraded to the latest firmware versions as of like a week ago. I dumped the stock Lenovo BIOS using a Raspberry Pi and ran ME cleaner on it and flashed it back as usual with flashrom. Well, it wouldn't boot at all. Nothing. No lights, no beeps, just dead. So then I restored the stock BIOS and switched around a couple of settings where I turned the security chip back on to "active", enabled all devices like cameras and mics, then turned off wake on LAN. Those were all of the settings that I touched. Basically I was just toggling stuff to see if it would change anything. I then dumped the BIOS again and ran it through ME cleaner and flashed it back. This time for whatever reason it booted right away. Under Ubuntu the ME no longer shows up under PCI devices. 

**An issue I encountered:**

So there's the good news but I also have bad news. The bad news is that sometimes on boot the machine gets to the splash screen and then I get a blank black screen where it hangs indefinitely until I force it to shut off by holding the power button. I'm not sure if this is an issue with the BIOS or the ME firmware but it happens maybe every 4-8 startups. It never happens when I reboot from the OS. It only happens when I power off and then boot the machine. I consider this a minor inconvenience, but still noteworthy. 

**Possible solution to the boot hang problem:**

If it's a BIOS issue causing the machine to hang on a black screen at boot, then flashing an alternative BIOS could fix the problem. The X230 does have Coreboot support (https://www.coreboot.org/Board:lenovo/x230), with a very recent build listed on the page of supported boards. I'm thinking about testing it out if I can find another used, working X230 for less than $100 on ebay. Since I really like using this machine I don't want to mess with this one more than I have to. 

**I probably messed it up:**

Not sure if toggling BIOS settings did anything. I think I might have just had bad contact with the clip to the ROM chip the first time and it screwed up the dumps. If that's the case it must have been reading out consistently bad images because all three of the first dumps matched in size and md5 hash. 

**Hardware specs:**

-Intel Core i5 3320M 2.6GHz
-Intel ME Windows driver 11.0.0.1157 from 7/7/15 (before flash, doesn't show now) 
-Trusted Platform Module 10.0.14393.479 from 6/21/06 (still shows after flash)
-Lenovo BIOS G2ETA8WW (2.68) from 4/19/17

[image]

corna, 2018-01-10:
It seems to me more like a BIOS bug.

I have me_cleaner on my X220 (which should be very similar to the X230, Ivy Bridge is just the "tick" of Sandy) with coreboot, and I've never experienced any major bug related to ME. I really suggest you to install coreboot and see if it fixes the issues, it should work wonderfully on your laptop ([stable native raminit](https://www.coreboot.org/Intel_Native_Raminit#Sandybridge.2FIvybridge) and [fully reverse-engineered EC](https://github.com/coreboot/coreboot/tree/master/src/ec/lenovo/h8)).

The recent [boot reports](https://review.coreboot.org/cgit/board-status.git/log/lenovo/x230) have been done by @persmule, maybe he can tell you the current status on this laptop.

matt123b, 2018-01-10:
@corna Ok so I'm back I have some more news. I restored to the backup of the stock BIOS that I had made, which was version 2.68. Ever since I forgot to check the EFI versions with the MacBook I've been more observant of these things. So I restored the old image and then from Windows I updated the BIOS to version 2.70 which came out late last year, and it worked fine using the official Lenovo utility. I then repeated the above process basically and tried ME cleaner again, and the boot hang issue is gone, and under Ubuntu the ME is gone from the PCI devices. So with confidence I can say that I've narrowed it to something with the BIOS, specifically version 2.68, and it has been fixed in 2.70. The Lenovo docs also talk about rollback protection on the BIOS, so I don't think I could test with other versions (not that it would matter since it's fixed). I might try out Coreboot if I have some spare time, but for now this seems to be resolved.

corna, 2018-01-10:
Great!

closed 2018-01-10 by corna