corna/me_cleaner #135: Disable ME by setting HDA_SDO pin high?

probonopd, 2017-12-20:
By searching around, I find references to disabling ME by setting the HDA_SDO pin high at boot, which is supposed to allow full SPI flashing. Does anyone have more information on/experience with this?

skochinsky, 2017-12-20:
It works similar to HAP/AltMeDisable (ME checks the pin on startup and halts if it's high), except in addition it overrides all flash descriptor permissions (SDO=SPI Descriptor Override), giving the host (CPU) access to the full flash (including ME and descriptor regions).

probonopd, 2017-12-20:
Thanks @skochinsky. Does this mean that by setting the `HDA_SDO` pin high at boot one can flash a new firmware from the running system using `flashrom` without having to access the flash chip's SPI lines physically?

corna, 2017-12-21:
There may still be issues with `flashrom` (like the vendor firmware setting a restricted set of opcodes, different from the ones required by flashrom) but technically yes.

closed 2018-01-07 by corna