corna/me_cleaner #137: how to check if me_cleaner did the job

Mafeuser, 2017-12-27:
I've flashed coreboot rom onto my x220 twice.
first version was with original flashregion_2_intel_me.bin (taken from latest vendor BIOS).
second version with flashregion_2_intel_me.bin modified by me_cleaner (no parameters given to me_cleaner).

here is the result when coreboot rom with original flashregion_2_intel_me.bin is running (Note "failed to become ready" in the end. why?):
```
        $ sudo ./intelmetool -s
        Bad news, you have a `QM67 Express Chipset Family LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

        MEI found: [8086:1c3a] 6 Series/C200 Series Chipset Family MEI Controller #1

        ME Status   : 0x1e000255
        ME Status 2 : 0x60000006

        ME: FW Partition Table      : OK
        ME: Bringup Loader Failure  : NO
        ME: Firmware Init Complete  : YES
        ME: Manufacturing Mode      : YES
        ME: Boot Options Present    : NO
        ME: Update In Progress      : NO
        ME: Current Working State   : Normal
        ME: Current Operation State : M0 with UMA
        ME: Current Operation Mode  : Normal
        ME: Error Code              : No Error
        ME: Progress Phase          : Host Communication
        ME: Power Management Event  : Clean Moff->Mx wake
        ME: Progress Phase State    : Host communication established

        ME: Extend SHA-256: c20401a7f4baa166f69c313455db4ed508fd484bf2b89b42eba9a2395cb0f205

        ME: failed to become ready
        ME: failed to become ready
        ME: GET FW VERSION message failed
        ME: failed to become ready
        ME: failed to become ready
        ME: GET FWCAPS message failed

        $ sudo ./intelmetool -m
        Bad news, you have a `QM67 Express Chipset Family LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

        MEI found: [8086:1c3a] 6 Series/C200 Series Chipset Family MEI Controller #1

        ME Status   : 0x1e000255
        ME Status 2 : 0x60000006

        ME: FW Partition Table      : OK
        ME: Bringup Loader Failure  : NO
        ME: Firmware Init Complete  : YES
        ME: Manufacturing Mode      : YES
        ME: Boot Options Present    : NO
        ME: Update In Progress      : NO
        ME: Current Working State   : Normal
        ME: Current Operation State : M0 with UMA
        ME: Current Operation Mode  : Normal
        ME: Error Code              : No Error
        ME: Progress Phase          : Host Communication
        ME: Power Management Event  : Clean Moff->Mx wake
        ME: Progress Phase State    : Host communication established

        ME: Extend SHA-256: c20401a7f4baa166f69c313455db4ed508fd484bf2b89b42eba9a2395cb0f205

        ME: failed to become ready
        ME: failed to become ready
        ME: GET FW VERSION message failed
        ME: failed to become ready
        ME: failed to become ready
        ME: GET FWCAPS message failed
```

Now the result when coreboot rom keeps flashregion_2_intel_me.bin modified by me_cleaner.
```
         $ sudo ./intelmetool -s
         Bad news, you have a `QM67 Express Chipset Family LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

         MEI found: [8086:1c3a] 6 Series/C200 Series Chipset Family MEI Controller #1

         ME Status   : 0x1e003052
         ME Status 2 : 0x10320002

         ME: FW Partition Table      : OK
         ME: Bringup Loader Failure  : NO
         ME: Firmware Init Complete  : NO
         ME: Manufacturing Mode      : YES
         ME: Boot Options Present    : NO
         ME: Update In Progress      : NO
         ME: Current Working State   : Recovery
         ME: Current Operation State : M0 with UMA
         ME: Current Operation Mode  : Normal
         ME: Error Code              : Image Failure
         ME: Progress Phase          : BUP Phase
         ME: Power Management Event  : Clean Moff->Mx wake
         ME: Progress Phase State    : M0 kernel load

         ME: Extend SHA-256: 0311dee9ea06e2ed7a972729ddbefd07e4406ed70f136e3bcca940fa0b705e13

         ME: has a broken implementation on your board withthis firmware
         ME: failed to become ready
         ME: failed to become ready
         ME: GET FW VERSION message failed
         ME: failed to become ready
         ME: failed to become ready
         ME: GET FWCAPS message failed
```

and "cbmem -c" result. (why "Current Operation State" is set to Normal?)

```
       $ sudo ./util/cbmem/cbmem -c | grep ^ME
       ME: FW Partition Table      : OK
       ME: Bringup Loader Failure  : NO
       ME: Firmware Init Complete  : NO
       ME: Manufacturing Mode      : NO
       ME: Boot Options Present    : NO
       ME: Update In Progress      : NO
       ME: Current Working State   : Recovery
       ME: Current Operation State : Bring up
       ME: Current Operation Mode  : Normal
       ME: Error Code              : No Error
       ME: Progress Phase          : BUP Phase
       ME: Power Management Event  : Clean Moff->Mx wake
       ME: Progress Phase State    : 0x4e
       ME: FWS2: 0x104e0002
       ME:  Bist in progress: 0x0
       ME:  ICC Status      : 0x1
       ME:  Invoke MEBx     : 0x0
       ME:  CPU replaced    : 0x0
       ME:  MBP ready       : 0x0
       ME:  MFS failure     : 0x0
       ME:  Warm reset req  : 0x0
       ME:  CPU repl valid  : 0x0
       ME:  (Reserved)      : 0x0
       ME:  FW update req   : 0x0
       ME:  (Reserved)      : 0x0
       ME:  Current state   : 0x4e
       ME:  Current PM event: 0x0
       ME:  Progress code   : 0x1
       ME: FWS2: 0x102c0002
       ME:  Bist in progress: 0x0
       ME:  ICC Status      : 0x1
       ME:  Invoke MEBx     : 0x0
       ME:  CPU replaced    : 0x0
       ME:  MBP ready       : 0x0
       ME:  MFS failure     : 0x0
       ME:  Warm reset req  : 0x0
       ME:  CPU repl valid  : 0x0
       ME:  (Reserved)      : 0x0
       ME:  FW update req   : 0x0
       ME:  (Reserved)      : 0x0
       ME:  Current state   : 0x2c
       ME:  Current PM event: 0x0
       ME:  Progress code   : 0x1
       ME: Requested BIOS Action: Continue to boot
       ME: FW Partition Table      : OK
       ME: Bringup Loader Failure  : NO
       ME: Firmware Init Complete  : NO
       ME: Manufacturing Mode      : NO
       ME: Boot Options Present    : NO
       ME: Update In Progress      : NO
       ME: Current Working State   : Recovery
       ME: Current Operation State : Bring up
       ME: Current Operation Mode  : Normal
       ME: Error Code              : No Error
       ME: Progress Phase          : BUP Phase
       ME: Power Management Event  : Clean Moff->Mx wake
       ME: Progress Phase State    : 0x2c
       MEBASE 0xfe000000
```

I'm confused, ie. I don't know if me_cleaner did the job or not?

Could You help with resolving the puzzle?

regards.

zamaudio, 2017-12-28:
The key here is the progress state of the ME boot :  `ME: Progress Phase          : BUP Phase`
And also `ME: has a broken implementation on your board with this firmware`
This means the ME is crippled.

Mafeuser, 2017-12-28:
@zamaudio, thank You for clarification

closed 2018-01-10 by corna