corna/me_cleaner #138: Not working with Gigabyte GA-B250M-D3H

seropp, 2017-12-29:
As others have pointed out for older but similar motherboards, under the best circumstances, it reaches the point where the bios would start loading the OS, then gets stuck in a reboot loop.

The Q-flash utility has some sort of checksum, you can bypass that problem by using the modded efiflash utility, running under freedos.

Flashrom does not detect the chipset. I don't have an external tool to flash this.

My guess is that somebody would need to do find the chunk of bios code that runs just before loading the OS and do something nasty to it, as it is doing to us, but I am not able to do that.

Thank you for the wonderful tools! (I might change the motherboard soon...)

nagyrobi, 2018-01-05:
Same motherboard here. Sad it bricks it...

nagyrobi, 2018-01-05:
idftool on the image file coming from factory (already on the board, but also downloadable, version F6):
`$ ./ifdtool -d B250MD3H.F6
File B250MD3H.F6 is 8388608 bytes
FLMAP0:    0x00040003
  NR:      0
  FRBA:    0x40
  NC:      1
  FCBA:    0x30
FLMAP1:    0x5a100208
  ISL:     0x5a
  FPSBA:   0x100
  NM:      2
  FMBA:    0x80
FLMAP2:    0x00310330
  PSL:     0x3103
  FMSBA:   0x300
FLUMAP1:   0x00000adf
  Intel ME VSCC Table Length (VTL):        10
  Intel ME VSCC Table Base Address (VTBA): 0x000df0

ME VSCC table:
  JID0:  0x00184fef
    SPI Componend Device ID 1:          0x18
    SPI Componend Device ID 0:          0x4f
    SPI Componend Vendor ID:            0xef
  VSCC0: 0x20052005
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x50
    Lower Write Status Required:        No
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    4KB
    Upper Erase Opcode:                 0x20
    Upper Write Enable on Write Status: 0x50
    Upper Write Status Required:        No
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    4KB
  JID1:  0x001840ef
    SPI Componend Device ID 1:          0x18
    SPI Componend Device ID 0:          0x40
    SPI Componend Vendor ID:            0xef
  VSCC1: 0x20252025
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x50
    Lower Write Status Required:        No
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    4KB
    Upper Erase Opcode:                 0x20
    Upper Write Enable on Write Status: 0x50
    Upper Write Status Required:        No
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    4KB
  JID2:  0x001860ef
    SPI Componend Device ID 1:          0x18
    SPI Componend Device ID 0:          0x60
    SPI Componend Vendor ID:            0xef
  VSCC2: 0x20252025
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x50
    Lower Write Status Required:        No
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    4KB
    Upper Erase Opcode:                 0x20
    Upper Write Enable on Write Status: 0x50
    Upper Write Status Required:        No
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    4KB
  JID3:  0x001820c2
    SPI Componend Device ID 1:          0x18
    SPI Componend Device ID 0:          0x20
    SPI Componend Vendor ID:            0xc2
[... 132 lines cut ...]
  EC Region Write Access:            enabled
  Platform Data Region Write Access: enabled
  GbE Region Write Access:           enabled
  Intel ME Region Write Access:      enabled
  Host CPU/BIOS Region Write Access: enabled
  Flash Descriptor Write Access:     enabled
  EC Region Read Access:             enabled
  Platform Data Region Read Access:  enabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       enabled
  Host CPU/BIOS Region Read Access:  enabled
  Flash Descriptor Read Access:      enabled
Found Processor Strap Section
????:      0x00000000
????:      0x80118012
????:      0x00000295
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff`

GigaByte released BIOS versions F7 and F8 for this board:
https://www.gigabyte.com/Motherboard/GA-B250M-D3H-rev-10#support-dl
Note how F8 changelog states "Update intel ME for security vulnerabilities" LOL.

nagyrobi, 2018-01-05:
For latest BIOS image version F8:
`$ ./ifdtool -d B250MD3H.F8
File B250MD3H.F8 is 8388608 bytes
FLMAP0:    0x00040003
  NR:      0
  FRBA:    0x40
  NC:      1
  FCBA:    0x30
FLMAP1:    0x5a100208
  ISL:     0x5a
  FPSBA:   0x100
  NM:      2
  FMBA:    0x80
FLMAP2:    0x00310330
  PSL:     0x3103
  FMSBA:   0x300
FLUMAP1:   0x00000adf
  Intel ME VSCC Table Length (VTL):        10
  Intel ME VSCC Table Base Address (VTBA): 0x000df0

ME VSCC table:
  JID0:  0x00184fef
    SPI Componend Device ID 1:          0x18
    SPI Componend Device ID 0:          0x4f
    SPI Componend Vendor ID:            0xef
  VSCC0: 0x20052005
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x50
    Lower Write Status Required:        No
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    4KB
    Upper Erase Opcode:                 0x20
    Upper Write Enable on Write Status: 0x50
    Upper Write Status Required:        No
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    4KB
  JID1:  0x001840ef
    SPI Componend Device ID 1:          0x18
    SPI Componend Device ID 0:          0x40
    SPI Componend Vendor ID:            0xef
  VSCC1: 0x20252025
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x50
    Lower Write Status Required:        No
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    4KB
    Upper Erase Opcode:                 0x20
    Upper Write Enable on Write Status: 0x50
    Upper Write Status Required:        No
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    4KB
  JID2:  0x001860ef
    SPI Componend Device ID 1:          0x18
    SPI Componend Device ID 0:          0x60
    SPI Componend Vendor ID:            0xef
  VSCC2: 0x20252025
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x50
    Lower Write Status Required:        No
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    4KB
    Upper Erase Opcode:                 0x20
    Upper Write Enable on Write Status: 0x50
    Upper Write Status Required:        No
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    4KB
  JID3:  0x001820c2
    SPI Componend Device ID 1:          0x18
    SPI Componend Device ID 0:          0x20
    SPI Componend Vendor ID:            0xc2
[... 130 lines cut ...]
  Flash Descriptor Read Access:      enabled
FLMSTR5:   0xffffff00 (EC)
  EC Region Write Access:            enabled
  Platform Data Region Write Access: enabled
  GbE Region Write Access:           enabled
  Intel ME Region Write Access:      enabled
  Host CPU/BIOS Region Write Access: enabled
  Flash Descriptor Write Access:     enabled
  EC Region Read Access:             enabled
  Platform Data Region Read Access:  enabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       enabled
  Host CPU/BIOS Region Read Access:  enabled
  Flash Descriptor Read Access:      enabled
Found Processor Strap Section
????:      0x00000000
????:      0x80118012
????:      0x00000295
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
mi@LaborKliens:~/Programs/me_cleaner/me_cleaner-master$ 
`

nagyrobi, 2018-01-05:
Update2: flashed BIOS image prepared with the -s kill switch, using the modded Efiflash utility from freedos. The system automatically rebooted, then power-cycled itself 2 times, and then booted up normally. Since intelmetool doesn't give any relevant information, I tried with Intel's official intel_sa00086.py, but that one still says the Intel ME is there in place. No wonders, since I only tried to use the killswitch - but unfortunately I have no other means to verify if I succeeded or not.

I guess there was no success because of 2 reasons:
- if the kill switch was active, presumably intel_sa00086.py wouldn't have to detect it
- the system power-cycled two times after I flashed the patched image. it only power-cycled once when I flashed the unmodified image. I guess their "DualBIOS" feature triggered in which restored the original image when first tried to boot - that's why it had to power-cycle twice (the time between the cycles was approximately equal to the time it took to efiflash).

seropp, 2018-01-06:
@nagyrobi , if you've got the time + disposition, there's beta bioses out there, I don't have the time on my hands at the moment, and I suppose the risk of bricking is slightly higher than usual.

Strangely, I don't get the power cycle + reflash thing going on, I had to manually reflash with qflash. It's interesting that the tool has an option to reflash the bios with or without the ME region, but I haven't been able to use that constructively.

Best of luck to you!

https://www.win-raid.com/t154f16-Tool-Guide-News-quot-UEFI-BIOS-Updater-quot-UBU.html
https://www.gigabyte.com/Support/Utility/Motherboard#mefw

nagyrobi, 2018-01-06:
The "GIGABYTE Intel ME Critical FW Update Utility" has no relevance here IMHO, it's only for boards which don't get complete BIOS updates anymore, ME can be updated separately. For this borad, BIOS version F8 already contains the updated ME, and this is confirmed by 

`$ sudo ./intel_sa00086.py 
INTEL-SA-00086 Detection Tool
Copyright(C) 2017, Intel Corporation, All rights reserved

Application Version: 1.0.0.152
Scan date: 2018-01-06 11:29:18 GMT

*** Host Computer Information ***
Name: LaborKliens
Manufacturer: Gigabyte Technology Co., Ltd.
Model: B250M-D3H
Processor Name: Intel(R) Core(TM) i7-7700 CPU @ 3.60GHz
OS Version: Ubuntu 17.10 artful (4.13.0-21-generic)

*** Intel(R) ME Information ***
Engine: Intel(R) Management Engine
Version: 11.8.50.3399
SVN: 3

*** Risk Assessment ***
Based on the analysis performed by this tool: This system is not vulnerable. It has already been patched.

For more information refer to the INTEL-SA-00086 Detection Tool Guide or the
  Intel Security Advisory Intel-SA-00086 at the following link:
  https://www.intel.com/sa-00086-support`

With previous BIOSes, the utility stated that the system is vulnerable.

nagyrobi, 2018-01-13:
Yet another BIOS update, F9b for this board (contains CPU microcode updates against Meltdown/Spectre):
https://www.gigabyte.com/Motherboard/GA-B250M-D3H-rev-10#support-dl

secremgithub, 2018-07-30:
Any workaround?

I'm thinking of flashing with a programmer. Any idea if it would trigger the Gigabyte bios protection mechanisms? I only want to set the HAP bit. Give me a suggestion.

link: https://wiki.gentoo.org/wiki/Sakaki%27s_EFI_Install_Guide/Disabling_the_Intel_Management_Engine

Meganium97, 2024-08-01:
possibly use PR #384